riscv/riscv-isa-manual #355: ambiguous meaning of "pc" in AUIPC

MartinGeisse, 2019-03-25:
Execution of an instruction increments the PC (e.g. by 4 for fixed 32-bit instruction encoding), and the description of AUIPC does not mention whether the "old" pc (i.e. the position of the AUIPC) or the "new" pc (i.e. the position of the next instruction) is used.

In comparison, the description of JALR hints toward the "old" pc. Since several instructions raise that question, it might be useful (and would be sufficient IMHO) to define once at the beginning what "pc" refers to (assuming, of course, that the meaning is currently the same for all instructions).

Suggestion: Add the following text at the end of the introduction of section 2.4 (Integer Computational Instructions), assuming that pc in all cases means the current instruction:

Some instructions refer to pc, the current program counter. Execution of the instruction increments or otherwise modifies the pc. The meaning of pc in these instructions is its value before that modification.

aswaterman, 2019-03-25:
It says "adds this offset to the {\tt pc} of the AUIPC instruction" - to me that is not ambiguous at all.

MartinGeisse, 2019-03-25:
No, it says "adds this offset to the pc, then ..."; however, I checked against the master branch of the repo and it seems that somebody has indeed found and fixed the ambiguity there. The change just wasn't released yet. Since I assume that the changes pending in the master branch are going to be released in the next version of the spec, I'd agree that we can close this ticket.

closed 2019-03-25 by MartinGeisse

aswaterman, 2019-03-25:
Ah, sorry, forgot this was a more recent change.