riscv/riscv-isa-manual #356: pc increment ambiguity with jumps and branches

MartinGeisse, 2019-03-25:
JAL and branches specify a value to add to the pc. However, execution of an instruction also increments the pc (e.g. by 4 for fixed 32-bit encoding). The specification does not state whether the latter happens for jumps or taken branches, that is, whether the jump or branch offset is relative to the JAL/branch or relative to the next instruction.

Current wording for JAL:
The jump and link (JAL) instruction uses the J-type format, where the J-immediate encodes a signed offset in multiples of 2 bytes. The offset is sign-extended and added to the pc to form the jump target address.

Suggestion in case JAL increments AND adds the offset:
The jump and link (JAL) instruction uses the J-type format, where the J-immediate encodes a signed offset in multiples of 2 bytes. The pc is first incremented to the next instruction as usual. Then, the offset is sign-extended and added to the pc to form the jump target address.

Suggestion in case JAL only adds the offset but does not increment:
The jump and link (JAL) instruction uses the J-type format, where the J-immediate encodes a signed offset in multiples of 2 bytes. The offset is sign-extended and added to the pc to form the jump target address. The usual increment of pc to the next instruction does not happen for JAL.

Current wording for branches:
All branch instructions use the B-type instruction format. The 12-bit B-immediate encodes signed offsets in multiples of 2, and is added to the current pc to give the target address.

Suggestion in case branches increment AND add the offset:
All branch instructions use the B-type instruction format. The 12-bit B-immediate encodes signed offsets in multiples of 2, and is added to the current pc (after incrementing to the next instruction after the branch) to give the target address.

Suggestion in case branches only add the offset but do not increment:
All branch instructions use the B-type instruction format. The 12-bit B-immediate encodes signed offsets in multiples of 2, and is added to the current pc (without incrementing to the next instruction after the branch) to give the target address.

aswaterman, 2019-03-25:
I do not concur that this needs clarification.  "Current" means "current," not "next."

aswaterman, 2019-03-25:
Perhaps the potential for confusion results from the use of the term "pc."  Another way of expressing this would be to say "added to the address of the branch/jump instruction", then the incrementing of the pc is removed from the picture.

MartinGeisse, 2019-03-25:
I agree that my proposed wording is far from optimal. Your suggestion sounds better to me (especially shorter and simpler, which is good).

commit 825c9395 (2019-03-25): Change "pc" to "address" for clarity

Resolves #356
closed 2019-03-25 by aswaterman