riscv/riscv-isa-manual #360: MSTATUS.MPV and MSTATUS.MTL bit position not specified for RV32

avpatel, 2019-04-05:
Currently MSTATUS.MPV and MSTATUS.MTL bit positions are only specified for RV64 but not for RV32.

aswaterman, 2019-04-14:
As the commentary says, we plan to put these fields in `mstatush` for RV32.  They'll be in the corresponding bit positions (their RV64 `mstatus` positions, minus 32 bits).

Let's leave this ticket open until we formally introduce the `mstatush` CSR layout.

avpatel, 2019-04-14:
Thanks Andrew.

At the moment, this missing info is blocking us to extend OpenSBI for hypervisor extension. In fact, we are already 100% code ready in Xvisor and OpenSBI.

It would be great if you can add the MSTATUSH CSR sooner.

aswaterman, 2019-04-14:
Is the issue just that you don't have the mstatush address?  In the current plan, it will be 0x310.

And you know the bit positions.  So, you should be able to proceed, even though the spec isn't updated yet.

avpatel, 2019-04-14:
Thanks, we will proceed with 0x310 address for MSTATUSH CSR. This will unblock us.

avpatel, 2019-07-16:
Closing this issue because spec has been updated.

closed 2019-07-16 by avpatel