riscv/riscv-isa-manual #363: Instruction Scheduling

Jack10001, 2019-04-06:
Hello！ I want to know how risc-v schedule instruction!  Can you explain it for me?

jim-wilson, 2019-04-09:
Instruction scheduling is more of a compiler issue than an ISA issue.  How instruction scheduling works depends on the compiler, as different compilers will have different infrastructure for instruction scheduling.  It will also depend on the target, as different targets will have different pipelines that work best with different instruction schedules.

You weren't very specific about what you are looking for.  If you want general info about instruction scheduling info, then you should probably consult a compiler textbook.  There is also a wikipedia page that you could look at
https://en.wikipedia.org/wiki/Instruction_scheduling

If you want info for a specific target, then get documentation for that target that has info on the pipeline, and instruction latencies.

If you want info for a specific compiler, then try looking at that compiler's documentation and/or source code.

Jack10001, 2019-04-10:
Thanks a lot!

closed 2019-04-10 by aswaterman