riscv/riscv-isa-manual #365: Meaning of "error"

pdonahue-ventana, 2019-04-08:
Section 4.3.1 ("Addressing and Memory Protection") says: "Note that failing to mark a global mapping as global merely reduces performance, whereas marking a non-global mapping as global is an error."

It is unclear what is meant by "error" in this context.  In general, the document should be clear when distinguishing software bugs from hardware-detected error conditions.  Since the former is almost certainly what is meant here, can the wording be changed to "...marking a non-global mapping as global is a software bug" or something similar?  

I notice a similar situation in 3.1.6.2: "In practice, such settings would almost always be an error"
I also suggest "software bug" here.

Other uses of the word "error" seem unambiguous where it is either a noun with an appropriate adjective (e.g., "hardware error") or it is an adjective modifying an appropriate noun (e.g., "error responses from slave devices").

Thanks,

-Paul

aswaterman, 2019-04-08:
In this case, it means “software did the wrong thing.” The vagueness is
because the hardware takes no corrective action.

aswaterman, 2019-04-08:
Thanks for the suggestions; hopefully the edits address your concerns.

closed 2019-04-08 by aswaterman

pdonahue-ventana, 2019-04-08:
Perfect.  Thank you.

commit 051ff712 (2019-04-08): Elucidate two uses of the word "error"

Resolves #365