riscv/riscv-isa-manual #368: Don't require that VM page tables be supported in I/O regions

jhauser-us, 2019-04-13:
I submit that implementations of page-based virtual addressing should not be required to support page tables located in I/O regions of memory.  If this is already the intention, I was unable to find where it is stated in the document.  I suggest editing the last sentence of step 2 in section 4.3.2, "Virtual Address Translation Process":

> If accessing *pte* violates a PMA or PMP check, or, optionally, if *pte* is not located entirely in main memory, raise an access exception corresponding to the original access type.

I would avoid dictating that PTEs must always be in main memory, partly because such a requirement could add an unnecessary extra cost to some implementations, and partly because I believe embedded cores should be free to support page tables located in ROM if they want to.

A platform standard could require support for page tables located in any *cacheable* memory region without conflicting with the text I propose.

aswaterman, 2019-04-13:
I think this is a reasonable proposal, and I agree that requiring page tables be in main memory is a step too far.

aswaterman, 2019-08-27:
> @aswaterman <https://github.com/aswaterman> A couple of comments:
>
>    -
>
>    I didn't see a change to the original text quoted by @jhauser-us
>    <https://github.com/jhauser-us>, so the existing restriction still
>    needs to be removed from that statement
>    -
>
>    Is there any reason why support for table walk accesses is optional
>    vs. mandatory for memory? The current text in the access type PMA requires
>    read, write, and execute (instruction fetch) for memory regions, so it
>    seems a bit inconsistent to make table walks optional via PMA. (Makes
>    perfect sense to do that for I/O regions.)

I don’t think JohnH’s proposed change is necessary. There’s already a
clause that PTE accesses check the PMAs, so simply making it clear that
walkability is a distinct PMA suffices.

Why not require cacheable main memory to be walkable? Krste pointed out
some secure embedded use cases where it is desirable to restrict where page
tables can live. We don’t see any reason to preclude those. Obviously you’d
never impose such a restriction on a Linux platform, and the platform spec
should require that all of cacheable main memory be walkable.

Reasonable?

aswaterman, 2019-08-27:
> To me, this seems like a protection function (specified in a PMP/PTE)
> rather than a capability function (specified in a PMA). For example, the
> architecture already requires that memory regions support instruction fetch
> accesses (but these can be protected by PMP/PTE settings), and one could
> make an argument that the same types of platforms would want to restrict
> where code lives. So in that sense, at least, page table accesses seem
> analogous.
> Granted, changing the PMA definition is easier than updating the PMP and
> PTE definitions. I suppose another way to get to "consistency" from my
> perspective is to update the access type PMA such that instruction fetch
> access support is optional for memory regions. Then, memory regions have to
> support explicit accesses (reads/writes), but they do not have to support
> implicit accesses (instruction fetches/page table accesses). Platforms
> could state that instruction fetches and page table accesses are required
> to memory regions (with the expectation that the Linux platform would
> require both).

I agree the same argument can be made for instruction fetch.