riscv/riscv-isa-manual #369: Can't trigger external interrupts to the Hypervisor when V=1

alistair23, 2019-04-16:
Currently section _5.2.5 Background Supervisor Interrupt Registers (bsip and bsie)_ has this note:

> [ NOTE: Need to describe how bsip.SEIP interacts with PLIC. Current thinking is
> that bsip.SEIP should purely be a read-write storage bit to emulate the PLIC for
> VS-mode; the PLIC should not be wired into bsip.SEIP. ]

Based on this note, I'm assuming that bsip.SEIP bit should never be set by an external interrupt. This leads to the following problem though:

Boot a bare metal Hypvervisor on a single core RISC-V CPU. At this point there is no guest booting so we are running in HS (V=0). The bare metal Hypervisor then presents the user with a console to enter commands.

The user uses the console to start a single guest. The Hypervisor starts the guest, so now we are running in VS (V=1). At this stage although the guest is running the user is still presented with the Hypervisor prompt, where they can for example stop the guest.

So far everything is fine.

To ensure that the Hypervisor still runs sometimes we set up a timer which will trigger and cause the Hypervisor to run. When the Hypervisor is woken up by the timer it checks for other pending interrupts (such as serial input from the user) and if nothing has happened it returns back to the guest in VS mode.

Here is the problem. Generally the Hypervisor will only wake for a small amount of time to check for pending operations. In that small amount of time it is unlikely that the user will be able to tripper an SEIP interrupt from entering serial input (or other external sources). This means that the Hypervisor will never see user input once the guest has started running.

What instead happens is when the user enters input it triggers the V=1 sip.SEIP bit which sends an interrupt to the guest (based on sie.SEIE). As the bsip.SEIP bit is not set (according to the note) it is impossible for the Hypervisor to know that an external interrupt occurred. This makes it impossible to enter serial input to the Hypervisor, locking the user out of the system.

This results in the Hypervisor having to do one of the two:

1. Poll the serial device when the hardware timer triggers to give the user a chance to enter text.
2. When HS wakes up check the current bsip.SEIE bit status and then take action based on that.

It might be worth wiring bsip.SEIP to the PLIC if V=1 as then it allows both the Hypervisor and the guest to accept the interrupt. In this way the Hypervisor can act on the interrupt first, which would allow it to clear the guest sip.SEIP (bsip when V=0) bit if required. The downside of this is excessive context switching if we are sending interrupts to the guest only.

Maybe something a future PLIC can be better suited to delegating?

alistair23, 2019-04-17:
@atishp04 @avpatel

jhauser-us, 2019-04-17:
Andrew could've been a little more careful with his note.  Instead of `bsip`.SEIP, I believe what he means is this:
> Current thinking is that the VS-level `sip`.SEIP should purely be a read-write storage bit to emulate the PLIC for VS-mode; the PLIC should not be wired into the VS-level `sip`.SEIP.

When Andrew wrote `bsip`, he was thinking of where the VS-level `sip`.SEIP is located when V = 0.  But, as you know, when V = 1, the foreground and background `sip` CSRs swap places.

> What instead happens is when the user enters input it triggers the V=1 sip.SEIP bit which sends an interrupt to the guest (based on sie.SEIE).

The behavior you describe is not what we intend.  To be fair, you'd be hard-pressed to divine this in the current documentation, since very little has been written for how interrupts should interact with the new VS and VU modes.

SiFive's PLIC can ordinarily be configured to direct some interrupts to M mode and some to S mode.  When the hypervisor extension is supported, simply replace "S mode" with "HS mode".  In his note, Andrew is postulating that the PLIC does not add an ability to direct interrupts to VS mode.

For your example, the console interrupts will always be directed to HS mode.  The hart's current virtualization mode (V = 0 or 1) makes no difference.

It is a perfectly good question to ask whether the PLIC *should* be able to direct interrupts to VS mode.  We know that the arrangement I describe, where all guest interrupts are routed through the hypervisor, must be supported.  Beyond that observation, I leave to another place and time any discussion about sending interrupts directly to a guest.

alistair23, 2019-04-17:
Ok, that makes sense to me. If all the interrupts are directed to HS mode SIP register (no matter the current state) then the example I described above works.

Can we ensure that at least the documentation updates to make that clear.

Ideally as you mentioned it would be nice to have some control over interrupts directly to the guest, as trapping back to the Hypvervisor every time is slow.

aswaterman, 2019-04-20:
Thank you, @jhauser-us -- I had intended for that note to be eliminated in favor of something more elucidatory, but clearly that never happened.  For the time being, I'll replace it with what you wrote, plus an explanation of our intent to specify a hardware-virtualized PLIC sometime down the road.

closed 2019-06-17 by aswaterman