riscv/riscv-isa-manual #378: How is global mapping (G bit in a PTE) supposed to work?

lbmeng, 2019-04-30:
Current privileged spec says:

`The G bit designates a global mapping. Global mappings are those that exist in all address spaces.
For non-leaf PTEs, the global setting implies that all mappings in the subsequent levels of the page
table are global. Note that failing to mark a global mapping as global merely reduces performance,
whereas marking a non-global mapping as global is a software bug that, after switching to an
address space with a diﬀerent non-global mapping for that address range, can unpredictably result
in either mapping being used.

Global mappings need not be stored redundantly in address-translation caches for multiple
ASIDs. Additionally, they need not be ﬂushed from local address-translation caches when an
SFENCE.VMA instruction is executed with rs2̸=x0.`

Based on the spec, assume we have the following setup:

* A PTE table is configured with several pages marked as global mapping under ASID x, and its PPN is written to satp.
* A new PTE table is configured with additional PTEs that map additional non-global memory under ASID y, and its PPN is written to satp followed by sfence.vma x0, <asid y>

For the global mappings that have already been cached in the TLB, in the new ASID y context, access to these memory mapped by the cached global mappings should work as expected. But for those global mappings that have not been cached anywhere in the TLB in previous ASID x context, how does the processor know these mappings when trying to access to these memory?

nbdd0121, 2019-04-30:
I don't quite understand what's the setup you're mentioning: `sfence.vma x0, x0` means flush all translation caches. Do you mean `rs2 != x0`?

You'll need to make sure all global pages are mapped with consistent meanings in all page tables.

aswaterman, 2019-04-30:
As @nbdd0121 said, global mappings must be stored redundantly in each page table.

lbmeng, 2019-05-01:
> I don't quite understand what's the setup you're mentioning: `sfence.vma x0, x0` means flush all translation caches. Do you mean `rs2 != x0`?

Sorry, this is a typo. Yes, I mean `rs2 != x0`

> You'll need to make sure all global pages are mapped with consistent meanings in all page tables.

OK, with such setup I can understand how it works. Originally I thought it was a mechanism of software optimization that global mappings can exist in just one page table hence software is not required to provide global mappings in all page tables, but that is wrong based on this information you provide.

So the value of G bit in a PTE is only a hardware optimization, if such a PTE is cached by hardware, sfence.vma x0, rs2 (!=x0) can leave it in the translation caches.

Is my understanding correct?

lbmeng, 2019-05-01:
> As @nbdd0121 said, global mappings must be stored redundantly in each page table.

Thanks for the confirmation. I believe it's worth mentioning such statement explicitely in the spec.