riscv/riscv-isa-manual #385: Document all custom instruction encodings

jhauser-us, 2019-05-17:
A reliable source informed me earlier this year that instructions in the SYSTEM major opcode with these binary encodings are available for custom use by RISC-V implementations:

    010111xxxxxxxxxxx000xxxxx1110011
    011011xxxxxxxxxxx000xxxxx1110011
    011111xxxxxxxxxxx000xxxxx1110011
    1000xxxxxxxxxxxxx000xxxxx1110011
    100111xxxxxxxxxxx000xxxxx1110011
    101011xxxxxxxxxxx000xxxxx1110011
    101111xxxxxxxxxxx000xxxxx1110011
    110011xxxxxxxxxxx000xxxxx1110011
    110111xxxxxxxxxxx000xxxxx1110011
    111011xxxxxxxxxxx000xxxxx1110011
    111111xxxxxxxxxxx000xxxxx1110011

For example, Rocket uses encoding

    111111000000 rs1 000000001110011

for its nonstandard CFLUSH.D.L1 instruction.

The list above is unofficial and may be incorrect.  The complete set of encodings available for custom instructions should be documented for everyone to know.

aswaterman, 2019-05-17:
Deepthroat's insight is that the upper 12 bits of these encodings exactly match the CSR address space that's reserved for custom use: https://github.com/riscv/riscv-isa-manual/blob/2bc4e587be2368657d34afbbf7c83fd1fc924dcb/src/priv-csrs.tex#L50

jhauser-us, 2019-05-17:
Andrew makes a good point that I neglected to mention.  But, just so no one misunderstands, the encodings I listed above are *not* CSR instructions.  Hence, the custom CSR address space does not by itself imply that the instructions above are custom.

aswaterman, 2021-08-29:
I've written up a variant of this scheme, which doesn't avail quite so much encoding space for custom use, but does subsume the various instructions/spaces mentioned in this thread.  b68de1df0c7269b34725cd371e3d6798c4c6fe1d

closed 2021-08-29 by aswaterman