riscv/riscv-isa-manual #386: RV64 FENCE HINT instruction

luismarques, 2019-05-20:
The file `rv64.tex` says:

`All instructions that are microarchitectural HINTs in RV32I (see Section~\ref{sec:rv32i-hints}) are also HINTs in RV64I.`

Yet, unlike the other RV32 HINT instructions, `FENCE` is included in the RV32 table of HINT instructions but not in the RV64 table.

aswaterman, 2019-05-20:
That’s a bug. Care to make a pull request against the RV64 chapter fixing it?

luismarques, 2019-05-20:
> That’s a bug. Care to make a pull request against the RV64 chapter fixing it, by adding FENCE to the table?

Sure. Thanks.

luismarques, 2019-05-21:
> That’s a bug. Care to make a pull request against the RV64 chapter fixing it, by adding FENCE to the table?

I created a pull request but let me just point out another discrepancy. In the RVC documentation for each instruction the HINT variants are described, while for RV32 and RV64 that's not the case. Whether that's worth revising I leave it up to you.

aswaterman, 2019-05-21:
Thanks. We’re aware of the discrepancy and haven’t decided whether to do
anything about it. RVC is something of a special case because the encoding
is awkward and confusing, with instructions embedded within other
instructions, so we were more explicit in the text.

luismarques, 2019-05-23:
Closed by #387.

closed 2019-05-23 by luismarques

abdullahyildiz, 2020-04-29:
What is the mnemonic for RV64/RV32 FENCE HINT instruction?

For now, I can use something like

`.4byte 0x0000000f`

in assembly code, but I am not sure if I'm doing it right.