riscv/riscv-isa-manual #389: C.NOP/C.ADDI HINT documentation

luismarques, 2019-05-23:
There is an inconsistency in the way that the C.NOP/C.ADDI HINTs are documented that IMO is confusing. Here's an example of the typical way that compressed HINT instructions are documented:

> C.LI is only valid when rd != x0; the code points with rd = x0 encode HINTs.

Contrast this with C.NOP/C.ADDI:

> C.ADDI is only valid when rd != x0. The code point with both rd=x0 and nzimm=0 encodes the C.NOP instruction; the remaining code points with either rd=x0 or nzimm=0 encode HINTs.

Given that that paragraph is encountered in a section that documents `C.ADDI*` instructions, that kind of sounds like rd=0 and nzimm=0 encodes a single C.NOP while rd=x0 *or* nzimm=0 encode `C.ADDI` HINTS. The section should probably be more explicit about which are `C.NOP` hints and which are `C.ADDI` hints, IMO.

jim-wilson, 2019-05-23:
There is an related unresolved binutils question about this, as it isn't clear on the binutils side what exactly in an instruction and what exactly is an alias here.
https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/18ObEKnn4mE

aswaterman, 2019-06-26:
I've changed the text to make it clear which hints are ADDI hints and which are NOP hints.

commit 9f0e2341 (2019-06-26): Clarify which hints are C.NOP hints and which are C.ADDI hints

Closes #389
closed 2019-06-26 by aswaterman