riscv/riscv-isa-manual #390: Replace the hypervisor's CSR swapping with a "substitution" model

jhauser-us, 2019-05-24:
The current draft of the hypervisor extension specifies that foreground and background supervisor CSRs are swapped whenever the virtualization mode changes (V changes from 0 to 1, or vice versa).  This swapping model has been criticized as unnecessarily complicating reasoning about the security of the system, since any mistake in fully swapping the CSRs at exactly the right time might allow lower-privileged code (VS mode) to determine the value of higher-privileged CSRs (HS mode).

An alternative model based on the concept of "substitution" of supervisor CSRs is believed to be easier to understand and prove correct.  The changes proposed are outlined as follows:

In the CSR address space, replace all of the background supervisor CSRs, `bs*`, with "VS CSRs" having names `vs*`.  When V = 0, these registers are proposed to be like the current `bs*` CSRs, holding the values of the VS-level supervisor CSRs.  And just as before, these CSRs would be directly accessible only in M mode or HS mode, not in VS mode.

However, unlike the current `bs*` CSRs, the `vs*` CSRs are proposed to be fully fleshed out to appear like the supervisor CSRs do in VS mode.  For example, when the N extension is implemented, `vsstatus` should include `ustatus` as a subset the same as `mstatus` and `sstatus` do.  And `vsstatus` should have read-only fields SD and XS that reflect what is relevant to VS level.  These fields are not in `bsstatus` currently because `bsstatus` is defined to hold the parts of `sstatus` that get swapped on a change into or out of the virtual machine mode.  The `vsstatus` CSR, on the other hand, can be defined more directly as "what VS mode sees in `sstatus`".

Lastly, when the virtualization mode changes (from V = 0 to 1, or vice versa), there would no longer be swapping of fields between the normal supervisor CSRs and background supervisor CSRs.  Instead, when V = 1, the defined VS CSRs would *substitute* for the corresponding supervisor CSRs.  Here's a draft of possible text:

> When V = 1, the VS CSRs substitute for the corresponding supervisor CSRs, taking over all functions of the usual supervisor CSRs except as specified otherwise.  Instructions that normally read or modify a supervisor CSR shall instead access the corresponding VS CSR.  In VS-mode, an attempt to read or write a VS CSR directly by its own separate CSR address causes an illegal instruction exception.
>
> While V = 1, the normal HS-level supervisor CSRs retain their values but do not affect the behavior of the machine unless specifically documented to do so.  Conversely, when V = 0, the VS CSRs do not ordinarily affect the behavior of the machine other than being readable and writable by CSR instructions.
>
> A few standard supervisor CSRs (`sedeleg`, `sideleg`, and `scounteren`) have no matching VS CSR.  These supervisor CSRs continue to have their usual function and accessibility even when V = 1.  Hypervisor software is expected to manually swap the contents of these registers as needed.

This proposal also requires changing the name of instruction HFENCE.BVMA to HFENCE.VVMA.

aswaterman, 2019-05-24:
I agree that this proposal is a bit easier to reason about than the current one.  For completeness, I'll also add that it's believed to be roughly cost-neutral.

aswaterman, 2019-05-30:
@atishp04 @alistair23 @avpatel Since you guys have extensive experience implementing the hypervisor extension, we would like your feedback on this proposal.  (We're pretty sure that it won't be too hard to change the software implementations to support it.)

avpatel, 2019-05-31:
Yes, this is lot easier to understand and cleaner approach.

From OpenSBI, Xvisor and KVM perspective, it will be just renaming of BS<xyz> to VS<xyz> but from QEMU perspective it will be bit more work (not a major functional change as such).

If we can have the hypervisor spec updated sooner then it would be great. We will update QEMU, OpenSBI, Xvisor, and KVM only after hypervisor spec is updated.

Also, along with this change can we also have "hypervisor guest time offset" and MSTATUSH defined in hypervisor spec.

Regards,
Anup

aswaterman, 2019-05-31:
Thanks for the feedback. I think we will have a new draft pretty soon, courtesy of @jhauser-us.

alistair23, 2019-06-09:
I remember when implementing the background registers in QEMU that the swapping was hard to comprehend and wasn't that clear, hence https://github.com/riscv/riscv-isa-manual/issues/369. So this sounds good to me. Hopefully it isn't too hard to update implementations on top of it.

aswaterman, 2019-06-16:
Thanks for the feedback, @alistair23.  We've updated the spec according to this proposal (b8e20ea56d3a83c115ae7e6ff90c9b966a837248); hopefully it isn't too painful for you, @avpatel, and @atishp04.

aswaterman, 2019-06-16:
One other note - some of the CSR addresses (hstatus, hedeleg, hideleg, hgatp) have changed to prevent a future non-orthogonality for custom CSR addresses.  They are included in that same commit.  I have also updated the riscv-opcodes repo: https://github.com/riscv/riscv-opcodes/commit/e7efefd3326e3e73d2063718820f2966113f0516#diff-28894eb6cc01cf77bef920641567df36R143

closed 2019-06-16 by aswaterman

aswaterman, 2019-06-16:
Oh, and the new HFENCE.VVMA has a different opcode from the old HFENCE.BVMA, for the same reason.

aswaterman, 2019-06-16:
Oh, and HFENCE.BVMA has been renamed to HFENCE.VVMA; and HFENCE.GVMA's opcode has been changed for the same reason the CSR addresses have changed.