riscv/riscv-isa-manual #399: interpreting OFF pmp entries

pmundkur, 2019-06-25:
I'm trying to understand this line in the PMP priority and matching logic section:

" If no PMP entry matches an S-mode or U-mode access, but at least one PMP entry is implemented, the access fails."

What is the expected behavior for an S/U access if the platform has PMP support and PMP registers, but they are all set to OFF?

I'd expect that access to succeed.  If this is the case, perhaps this line could be clarified to "... at least one PMP entry is enabled ..."  or something similar.

If the access is supposed to fail, then it would be good to explicitly add a line like: "In particular, if all PMP entries are OFF, an S-mode or U-mode access fails."

aswaterman, 2019-06-25:
It's the latter option.  I claim this follows from a literal reading of the nominal text, but I'll add a commentary note to prevent any confusion.

pmundkur, 2019-06-25:
That would be great, thanks.

commit 0b25db1f (2019-06-25): Clarify that, if all PMPs are OFF, all S/U accesses fail

Closes #399
closed 2019-06-25 by aswaterman