riscv/riscv-isa-manual #403: why the supervisor mode not define the stime/stimecmp register

amwyga, 2019-07-02:
the supervisor CSR register SIP/SIE defined the STIE/STIP bit that enables the supervisor level timer, and the scause register also defines the supervisor-level timer interrupt, but why does the supervisor mode not define the stime/stimecmp register?

aswaterman, 2019-07-02:
This paragraph explains the "why": https://github.com/riscv/riscv-isa-manual/blob/9f0e2341734fb3dcbcec737fd21ff8d5f4e6ea00/src/supervisor.tex#L9

As to the "how," for must current systems, the SBI provides hooks for that.  Future systems may expose these mechanisms directly to S-mode.

amwyga, 2019-07-02:
> 本段解释了“为什么”：
> 
> https://github.com/riscv/riscv-isa-manual/blob/9f0e2341734fb3dcbcec737fd21ff8d5f4e6ea00/src/supervisor.tex#L9
> 
> 至于当前系统的“如何”，SBI为此提供了钩子。未来的系统可能会将这些机制直接暴露给S模式。

thank you！

> This paragraph explains the "why":
> 
> https://github.com/riscv/riscv-isa-manual/blob/9f0e2341734fb3dcbcec737fd21ff8d5f4e6ea00/src/supervisor.tex#L9
> 
> As to the "how," for must current systems, the SBI provides hooks for that. Future systems may expose these mechanisms directly to S-mode.

thank you！

closed 2019-07-02 by amwyga