riscv/riscv-isa-manual #407: Questions on mtimecmp behaviour

gameboo, 2019-07-12:
Hi,

I am reading section `3.1.10 Machine Time Registers (mtime and mtimecmp)` as I am cleaning the code for a clint I have. A couple questions came to me as I was reading:

*  The section says:
> The interrupt remains posted until it is cleared by writing the `mtimecmp` register.

and later adds:
>Writes to `mtime` and `mtimecmp` are guaranteed to be reflected in MTIP eventually, but not necessarily immediately.

Should I clear the timer interrupt on writes to `mtime` as well as writes to `mtimecmp`? Or is this last sentence completely orthogonal to the actual clearing of the interrupt and actually referring to the fact that writing `mtime` may eventually trigger the comparison logic in the CLINT and in turn potentially be reflected via a later interrupt?

* How exactly does the previous sequence prevents an interrupt from happening after the `sw t0, mtimecmp` and before the end of the sequence? Say `mtimecmp` was `0x00000000ffffff00` or so and `mtime` reaches it, triggering the exception and eventually the `mtimecmp` update sequence. Say by the time we write `-1` in the bottom half of `mtimecmp`, `mtime` has reached `0x00000000fffffffe`. We now have `0x00000000ffffffff` in `mtimecmp`, which means that we won't be able to perform our following writes before getting an interrupt again. Am I missing something there? Unless maybe we expect that sequence to be ran with interrupts disabled?

*  Finally, in my RV32 implementation, I was (possibly naively?) trying to make sure to only write the underlying 64 bit register and clear the interrupt in an atomic manner, and therefore took the view that only the write to one of the two 32bit halves would trigger those side effects. In particular, I am buffering writes to the high half and folding them in and clearing the interrupt on writes to the low half. It seems to me that this implementation (which can be found [here](https://github.com/CTSRD-CHERI/RVBS/blob/master/src/CLINT.bsv#L107) for more details) behaves correctly with the suggested sequence:
```
# New comparand is in a1:a0.
li t0, -1
sw t0, mtimecmp # No smaller than old value.
sw a1, mtimecmp+4 # No smaller than new value.
sw a0, mtimecmp # New value.
```
Is it acceptable to have the side effects of a write to `mtimecmp+4` only ever visible after a write to `mtimecmp`, or am I in violation of the spec?

Thank you.

aswaterman, 2019-07-12:
MTIP is based solely on the comparator output. Writes to mtime/mtimecmp
set/clear the interrupt only if they cause the comparator’s output to
change.

There’s no guarantee a timer interrupt won’t occur during that code
sequence (unless of course the interrupt is masked). Its purpose is only to
prevent spurious interrupts due to the fact that it takes two stores to
write mtimecmp.

There is no requirement that both halves of the register be written in
RV32. Buffering half of the write until the other half arrives is actually
incorrect.

gameboo, 2019-07-12:
Thanks for the fast answer!

> MTIP is based solely on the comparator output. Writes to mtime/mtimecmp
> set/clear the interrupt only if they cause the comparator’s output to
> change.

May I suggest to explicitly add a mention of `mtime` in the first sentence? Maybe something like:
> The interrupt remains posted until it is cleared by a write to either the `mtime` or `mtimecmp` register.

> There’s no guarantee a timer interrupt won’t occur during that code
> sequence (unless of course the interrupt is masked). Its purpose is only to
> prevent spurious interrupts due to the fact that it takes two stores to
> write mtimecmp.

Sorry for being a bit dense here :grimacing:: can you explain what qualifies as a spurious interrupt here, if not the potential ones during the sequence (that could potentially separate the writes of the two halves)?

> There is no requirement that both halves of the register be written in
> RV32. Buffering half of the write until the other half arrives is actually
> incorrect.

Thanks I'll go and change my code.

aswaterman, 2019-07-12:
Yeah, I agree the first sentence should mention `mtime`.

The only intent of the -1 trick is to prevent mtimecmp from temporarily assuming a value that's smaller than `min(old value, new value)`.  By "spurious" I meant an interrupt that occurs as a result of mtimecmp temporarily falling below `min(old, new)`.

aswaterman, 2019-07-12:
https://github.com/riscv/riscv-isa-manual/commit/5960072e57140d9a56446e837d59fd5bf484434d

gameboo, 2019-07-12:
> The only intent of the -1 trick is to prevent mtimecmp from temporarily assuming a value that's smaller > than min(old value, new value). By "spurious" I meant an interrupt that occurs as a result of mtimecmp > temporarily falling below min(old, new).

Of course! That makes sense.
Do you reckon adding a note with exactly this would be useful (it certainly was to me! thanks again)?

5960072 looks good, and makes me realize that in my code, I was also explicitly latching the exception after the comparison went true, and clearing it on `mtimecmp` writes. The intention there was to avoid losing the interrupt on wrap around (the case I was thinking of here is that of someone writing a high value to `mtime` and to `mtimecmp` causing fast wrap around, as opposed to the one that consists in waiting for a 64-bit counter to wrap). Is the case of `mtimecmp` becoming greater than `mtime` as a result of a wrap around a case that is irrelevant?

aswaterman, 2019-07-12:
The implication of what I wrote is that `mtime` overflow might cause timer interrupts to fall.  I think that's perfectly acceptable (a primary reason to use a 64-bit counter is that it won't overflow unless software forces it to do so), and results in the simplest hardware implementation.

aswaterman, 2019-07-12:
(Note also this matches Spike's long-standing behavior: https://github.com/riscv/riscv-isa-sim/blob/master/riscv/clint.cc#L64)

gameboo, 2019-07-12:
Ok then, I'll go drop a latch from my RTL :) And yes I agree that is implied by what you wrote.
Thanks!

closed 2019-07-12 by gameboo

aswaterman, 2019-07-12:
Cool, I'll also add that note about the code sequence, as you suggested.