riscv/riscv-isa-manual #410: Preciseness of mtval information

jscheid-ventana, 2019-07-17:
For mtval (and *tval*), there’s a value proposition in the non-normative commentary (Section 3.1.17) in two places where “races” or “atomicity” are mentioned.  But it's not as clear in the normative content that this is a requirement.

Note that it might be difficult for some implementations to satisfy this atomicity constraint (e.g., with predecoded instruction caches).

aswaterman, 2019-07-17:
You mean, is it possible for mtval to be populated with an instruction other than the one that originally caused the illegal-instruction exception?  There is an implicit assumption that the instruction stuffed into mtval is the same one that caused the trap.  It is indeed not explicitly stated anywhere in the normative text... I guess we just thought it was obvious.

It would seem to me to be quite confusing for software to see an illegal instruction trap, but upon reading mtval, find an instruction that's actually legal.  (Of course, this is no worse than if this feature weren't implemented, and software were forced to go to memory to read the instruction.)

kasanovic, 2019-07-18:
The value in mtval has to be precise, otherwise supervisor could find
a different illegal but ok-to-emulate instruction in mtval, causing
strange behavior.

Krste

| You mean, is it possible for mtval to be populated with an instruction other
| than the one that originally caused the illegal-instruction exception? There
| is an implicit assumption that the instruction stuffed into mtval is the same
| one that caused the trap. It is indeed not explicitly stated anywhere in the
| normative text... I guess we just thought it was obvious.

| It would seem to me to be quite confusing for software to see an illegal
| instruction trap, but upon reading mtval, find an instruction that's actually
| legal. (Of course, this is no worse than if this feature weren't implemented,
| and software were forced to go to memory to read the instruction.)

| —
|

marceg, 2019-07-18:
Isn't a mismatch only possible between updating instruction memory and executing FENCE.I (or its eventual replacement) ?  Any attempt to execute instructions in the process of being updated between these two events sounds like a programming error.  If the architecture were to define single-copy atomicity on instruction updates, some limited updates might be possible, but given that breaks instruction translation between FENCE.I, that seems untenable.  It seems that ensuring atomicity might have a tiny chance to help debug such programming errors, but otherwise does not appear necessary?  In other words, if the trap handler finds a different instruction than expected in mtval, it'll result in strange behavior, but it's due to a programming error to begin with, so is this really an issue.

jscheid-ventana, 2019-07-18:
If the perspective for specification of function is focused on offloading exception handling SW, then the purpose of the mtval value is in emulating the instruction fetch that would otherwise have been performed by SW.  In which case atomicity isn't part of what it's offloading.

It's important to highlight what value is provided by the additional atomicity requirement, because for implementations that do not implement mtval, the additional SW work should be apparent.

My narrow intent with this issue is just to have the current intent clarified.

gfavor, 2020-09-06:
I think we all are agreeing that there is a form of atomicity requirement when mtval is loaded with the instruction being faulted, i.e. that it matches what was last actually fetched from memory (since the last FENCE.I).  (Although note that there still isn't any actual fetch atomicity requirement in the architecture, i.e that a misaligned 32-bit instruction must be fetched atomically.  It is just that whatever 32-bit instruction value a hart last fetched from memory must be what it decoded to cause this fault and must be what is recorded in mtval.  In other words, a self-consistency requirement.)

This means that an implementation with a decoded instruction cache must either be able to recreate the instruction that was previously fetched (for recording it into mtval), or it must (presumably) start over with the instruction and re-fetch and re-decode it from memory (and then base any trap action on that new instruction value).

As far as mtval off-loading or reducing instruction emulation overhead, it seems like the existing non-normative text makes it clear that without an mtval instruction value, the instruction emulation code must be prepared to deal with an atomicity issue when doing its own fetch of the instruction word.

David-Horner, 2020-09-06:
> This means that an implementation with a decoded instruction cache must either be able to recreate the instruction that was previously fetched (for recording it into mtval), or it must (presumably) [...]

A third alternative is also available, that the implementation sets mtval to zero.

SW may be just as confused, but the fault won't be hardware's.

allenjbaum, 2020-09-07:
Not that this issue is only for illegal instruction traps - all the others,
which return an address, shouldn't have this issue.

allenjbaum, 2020-09-07:
Oh, and (eventually) the compliance tests will expect the implementation to
provide a mask of which exception types set MTVAL appropriately (and the
illegal address mapping type ...)

It isn't clear to me whether a specific mcause can sometimes return zero,
and sometimes return the architecturally defined value, but I'm guessing
the answer is no; it must always be either one or the other.

David-Horner, 2020-09-07:
> Oh, and (eventually) the compliance tests will expect the
> implementation to
> provide a mask of which exception types set MTVAL appropriately (and the
> illegal address mapping type ...)
>
> It isn't [...]

 From the doc:

*The mtval register is an MXLEN-bit read-write register formatted as 
shown in Figure 3.26. When a trap is taken into M-mode, mtval is either 
set to zero or written with exception-specific information to assist 
software in handling the trap. Otherwise, mtval is never written by the 
implementation, though it may be explicitly written by software. **
*

/*The hardware platform will specify which exceptions must set mtval 
informatively and which may unconditionally set it to zero.*/

This appears to imply your guess. However, the wording "may 
unconditionally"  may leave an option for conditionally providing either.

A reasonable condition to allow conditionally zeroing appears to be when 
cached instruction is suspect and in memory instruction has changed.

allenjbaum, 2020-09-07:
I’m going to assert that the definition must be testable by a compliance test. 
Something that is different depending on microarchitectural state can’t be.

That is: compliance tests will be configured with implementation defined bitmasks which indicates which exceptions write MTVAL with addresses or instructions, and which with zero.

So if your implementation sometimes does it one way, and sometimes the other, you’re likely to fail compliance.

-Allen

David-Horner, 2020-09-07:
On Sun, Sep 6, 2020, 23:05 Allen Baum, <notifications@github.com> wrote:

> I’m going to assert that the definition

I infer that the definition to which you refer is the full Priviledge
Spec.and not just this situation.

must be testable

We already have specifications, such as allowed state at start up, that are
undefined and thus arguably 'not testable'.

> by a compliance test.
>
Yes, I agree that compliance testing is a critical consideration.
I don't agree that it must determine and restrict the specification.
Especially as you constrain the compliance testing:

> Something that is different depending on microarchitectural state can’t be
>
[allowed in a compliance test].

> At least that is how I read the assertion.

>
>
> That is: compliance tests will be configured with implementation defined
> bitmasks which indicates which exceptions write MTVAL with addresses or
> instructions, and which with zero.
>
That design may be insufficient for the spec.
I believe the spec is the dog in this tail wagging exercise.
There are other testing designs that can validate a profile that allows
zeros anytime, but when not zero must be the correct address or instruction
fragment.

>
> So if your implementation sometimes does it one way, and sometimes the
> other, you’re likely to fail compliance.
>
I suspect the specific situation we are discussing would yield a false
positive pass from the testing regimen as the fail situation would be very
difficult to manufacture.
If the situation were reliable testable then it can also validly be tested
as a condition which it is required to be zero by some profile and required
to give a definative but meaningless result in another profile.

> -Allen
>
> wrote:
> the
> zero,
> guessing

allenjbaum, 2020-09-07:
Yes, I'm referring to the spec wording.
Yes, the spec wording is a bit ambiguous. Compliance has an interpretation
that we will use. Implement something else and we will not guarantee that
you will pass, nor will we guarantee to change that failing test (or the
golden model...) to enable that you pass.

Yes, there are things that are undefined (e.g. exact reset values of GPRs
and most CSRs).
Compliance will not test architecturally defined undefined results. There
is no guarantee that an undefined opcode traps with an illegal opcode
exception -- so we don't test *any* undefined opcodes. There are some
opcodes that are guaranteed to trap. We can test that they do. Not
guaranteed - no test. (Note: there must be a way to indicate in a profile
that an opcode is guaranteed to trap, so you can emulate it if it isn't
supported in HW- compliance hasn't implemented that yet because profiles
haven't been defined yet)

We restrict compliance tests to only those features that are explicitly
defined by the spec.
As a special case, we may possibly test some WARL writes to CSRs if the
legal values have been restricted to the YAML syntax defined by the
compliance TG. We are arguing about testing implementation defined
illegal->legal mappings as well (as opposed to testing just that the result
of an illegal write is legal) if the mappings are also restricted to those
that compliance has defined.
If you try to implement something else, no guarantees that you will pass
compliance tests.
You can ask for a waiver if you do something else (waiver process TBD) but
you don't get to use the trademark if you don't pass compliance.

No,  to
       "allow conditionally zeroing ... when a cached instruction is
suspect and in memory instruction has changed"
is not reasonable, because e.g. the SAIL model knows nothing about caches -
that's a micro-architectural feature.
If, on the other hand, you can add support for all possible sizes and
configurations of caches to SAIL, SPIKE, and QEMU, we might think about it.
But probably not.

Our speced testing methodology is to compare your results to a golden
model. The golden model will be configurable up to a point - and if every
implementation gets to pick and choose arbitrary conditions for which MTVAL
is set zero, and for which it isn't - then we can't.
That is not supportable, so we won't support it.

Yes, false positives are possible. They are known as "bugs". Compliance is
not a substitute for DV. It is possible to pass compliance tests and be
buggy.
The purpose of compliance is to demonstrate that you have read, understood,
and implemented the spec. A corollary is that it tests that your DV is
testing the right thing and hasn't misinterpreted the spec

pdonahue-ventana, 2020-09-07:
At the risk of going off on a tangent (which I think has already happened):
The compliance suite methodology is not to do an instruction-by-instruction comparison with the golden model.  It just declares that certain values are interesting and stores them to the signature which later is compared.  The comment above assumes that the tests must do:
```
x1 = mtval
store x1 to signature
```

Why couldn't it do this which would work whether mtval is always zero on illegal instructions or this is a legal but non-deterministic one-time zero:
```
x1 = mtval
if x1 == expected_instruction then
  store 1 to signature
else if x1 == 0 then
  store 1 to signature
else
  store 0 to signature
endif
```
Then both the golden model and the DUT must have a signature of 1.

This check that tolerates all legal values is more along the lines of how they do it in other architectural compliance suites I've seen.  It's also the only way to handle things like the memory model since the formal model clearly cannot accurately model timing of an MP hardware platform.

This comment isn't really about the original question but is more about whether the architecture drives the compliance suite or the compliance suite drives the architecture.

allenjbaum, 2020-09-07:
Tests  don’t have an “expected value”.  The value is whatever it is, and we make no effort to predict it. 

That is the function of the golden model.

-Allen

pdonahue-ventana, 2020-09-07:
> Tests  don’t have an “expected value”.  The value is whatever it is, and we make no effort to predict it.

Tests certainly could have an expected value so why impose this limitation on all tests?  Especially if it means that either some tests cannot handle all legal values or that the architecture must change to conform to this artificially-imposed limitation?

allenjbaum, 2020-09-07:
The expected value is what the golden model produces, not something built
into the test. That's what we check against.
It was decided long ago (before my time) that we won't have self checking
tests, and what you are suggesting is a self checking test: return 0 if it
fails and 1 if it matches. We are not going there
What we are allowing is that any particular mcause can either always set
mtval according to the spec or always set it to 0.
That's my interpretation of the spec. Even if you could get Andrew to
declare otherwise (not impossible by any means), fixing that would require
augmenting the golden model for all (architectural) possibilities of what
people might implement.

To do that you would need to define a crisp architectural demarcation that
covers all the possibilities of how an implementation could decide to do it
one way or the other. That demarcation is currently set (per cause) as a
fixed 0 or 1.
Note that we have a single special case of HW address compression of MTVAL
(just for you, in fact) - based on a parameter defined by the model ( how
many bits are implemented in MTVAL). But, that isn't a case where we
sometimes compress and sometimes we don't, it is always compressed.

There is another possibility: don't do a direct comparison  but check that
the result is in the set of legal values on a mismatch (so the golden model
always saves the non-zero value, but the DUT could also save 0). This would
require re-working the framework to mark signature offsets where this is
possible, tie it to the CSR field(s) being checked, look each mismatched
field, map it to the YAML description of legal values for those fields, and
declare a match if the value the DUT stored was in the list of legal
values. We might do that someday, but in this particular case, (where HW
is writing the value, not SW) all values are potentially legal, so that
doesn't work.

aswaterman, 2020-09-08:
Architectural decision making obviously must incorporate verification and compliance-testing costs, but compliance doesn't get an unconditional veto.

It's straightforward enough to dynamically test for either 0 or the faulting instruction word that it doesn't seem appropriate to rule out, on the basis of compliance testing, the possibility that an implementation may return 0 sometimes and not other times.

Whether or not such an implementation will be too confusing for software to use is a separate matter.  (My take is that software can cope, because handling the 0 case dynamically in the same code path is the easiest way to write the code, and the runtime cost (a single, rarely taken branch) is low.)

aswaterman, 2020-09-08:
Now, to get back on topic...

I think we should resolve this issue by clarifying that, if mtval is written with the faulting instruction, it must be the instruction that was actually fetched.  This does make some implementations more expensive or complex, but they do have the choice of populating the register with 0 instead (either always, or only in sticky situations, like where the illegal instruction exception isn't easily detected during decode, as is the case for some vector instructions), so I think it is fair.

allenjbaum, 2020-09-08:
That isn’t the issue. It doesn’t affect the test at all, it affects the Sail model and the framework.
It’s not a simple change at all, it is quite pervasive. The framework has no idea what is being compared, just that it’s comparing a value. I listed all the changes ( that I know about) and it is nontrivial.

-Allen

aswaterman, 2020-09-08:
But Paul's suggestion works fine.  It's not adopting a self-checking methodology; it's just constraining the code generation so that when mtval is read for mcause = illegal instruction, the test program patches up the 0 case, e.g.,

    x1 = mtval
    if (x1 == 0) {
      x1 = *(uint16_t*)mepc
      if ((x1 & 3) == 3)
        x1 |= (*(uint16_t*)(mepc + 2)) << 16
    }

which is of course a simplified version of what an OS would do.  It's more code, but I don't think it's too much to ask...

allenjbaum, 2020-09-08:
I thought there were ugly corner cases, but the spec says that the entire
instruction must be fetched before an illegal op trap, I don't see any
issues with faulting in the handler (unclear if hypervisor would affect
that answer)
You need to save and set  the MPRV bit before the illegal opcode is
fetched, and restore it afterwards.
This is doable for illegal op traps, but not for any other cause that saves
an address sometimes, but not other times.

aswaterman, 2020-09-08:
The good news is that illegal-instruction exceptions are probably the only case where there's a justification for supplying 0 sometimes but not others.  In the other cases, zeroness being a static property is seemingly less limiting.

allenjbaum, 2020-09-08:
Ah, there is a corner case that I had forgotten about, and was actually the
main reason that the standard compliance trap handler works the way it does
(i.e.. we never decode the instruction length to skip past an op that
trapped - we always go to the next 8B aligned address from ePC+2) - it's
because we cannot guarantee that the code is readable if it is executable.
Fixing that is not always possible (think: locked PMP entry) and is ugly in
any case.

allenjbaum, 2020-09-08:
However - I think we can argue that compliance tests for illegal opcode
exceptions will never be run from memory that is executable but not
readable indeed, we need to be able to load the test, and get to decide the
environment it is run in), and that specific part of the handler will not
be executed unless it is an illegal opcode exception that returns zero in
mtval. The fact that  ucode can effectively be read (via mtval) even in
non-readable regions is a slight security hole, though hard to exploit,

aswaterman, 2020-09-08:
Yeah, it’s another reasonable constraint on the test programs.

jscheid-ventana, 2021-08-10:
> I think we should resolve this issue by clarifying that, if mtval is written with the faulting instruction, it must be the instruction that was actually fetched. This does make some implementations [...]

Can this move forward?  The longer this stays under-specified, the more chance there will be implementations that don't do this.

allenjbaum, 2021-08-10:
For what it's worth: the std trap handler in the arch-tests saves MTVAL,
but if it is an illegal op exception (xcause==2), then it will check to see
if it is zero.
IF it is, it will load set MPRV, 2 or 4 bytes of opcode pointed to by xepc,
and then restore MPRV  (I have a FIXME to update that if we ever get 6 or
8B instruction encodings).

aswaterman, 2021-08-11:
We should try to keep these threads on a single topic, and while the arch-tests matter is important, it is OT.

Back to the matter at hand: re-reading the spec, I don't see why we have to take any action to obtain the desired result.  The text says that mtval will be updated with "the actual faulting instruction".  No sensible reading of that phrase would permit some other instruction to be written to mtval instead.

closed 2021-08-11 by aswaterman

jscheid-ventana, 2021-08-12:
Thanks.  It's often the case that it's not clear that the specification is actually complete.  It this case my question is answered (that a class of implementations would not be compliant).