riscv/riscv-isa-manual #414: MISA bit definition for half-precision floating-point extension

chuanhua, 2019-07-22:
Could we define the MISA bit for half-precision floating-point extension using reserved K/O/R bits in MISA?

This bit is needed to indicate that if half-precision floating-point extension is present, then
1. If RVS is present, then FCVT.S.H and FCVT.H.S instructions will be added to RVS.
2. If RVD is present, then FCVT.H.D and FCVT.D.H instructions will be added to RVD.

aswaterman, 2019-07-22:
Perhaps this has changed, but IIRC, the plan was for half-precision support to be coupled to the vector unit.  So half-precision is present if (misa.V && misa.F).

chuanhua, 2019-07-26:
The above condition of "if (misa.V && misa.F) is not sufficient to indicate half-precision support is present. It is possible that half-precision is not supported, so when SEW is 16, and a floating-point vector instruction will generate an exception.

aswaterman, 2019-07-26:
That’s true.

chuanhua, 2019-08-07:
I narrow my MISA bit selection for half-precision floating-point extension to K because the number of 16 is related to a hexa**K**aidecagon.

aswaterman, 2019-08-07:
Ha!  I was not aware of that term; I have only heard hexadecagon.

I agree allocating a letter like K is a reasonable approach.

jscheid-ventana, 2019-08-07:
Is there an assumption about which half-precision format(s)?

-Josh

jim-wilson, 2019-08-07:
I did a little searching, and found that kai means "and" in greek, so hexakaidecagon is six and ten (poly)gon.  Still, it looks like a reasonable mnemonic to remember what K stands for.

aswaterman, 2019-08-07:
@jscheid-ventana IEEE 754.  But given interest in bfloat16, there might be some desire to support that, too, possibly with the same opcodes and an fcsr mode switch.

eak, 2019-08-19:
@aswaterman I would like to suggest two bits in the FCSR be added to indicate the format of 16-bit FP. 00 would IEEE, 01 would be bfloat16, and 10 would be some sort of 16-bit posit (since there seems to be some interest in posits, according to Krste). Of course an implementation is free to implement any subset of these if they implement 16-bit FP, trapping if the FCSR contains any unsupported value.

Comments?

aswaterman, 2019-08-19:
We had been thinking along the same lines.  (Note that for backwards compatibility, the field needs to be WARL, anyway, so the implementation can just refuse to write unsupported values, rather than trapping them.)

eak, 2019-08-19:
WARL sounds good. Bits 8-9 perhaps? Or would it be good to reserve bits for 16-bit, 32-bit, 64-bit, and 128-bit at the same time? That might suggest 16-17, 18-19, 20-21, 22-23.

aswaterman, 2019-08-19:
I don't think we need to support full associativity of format: it doesn't make much sense to be able to configure the machine such that the 16-bit mode is bfloat, the 32-bit mode is posit, and the 64-bit mode is IEEE, for instance.

In any case, all of the fcsr bits are reserved for standard use, so we don't need to pre-allocate them.  Along the same lines, we don't actually need to reserve space for the posit format until we specify such an extension.

pdonahue-ventana, 2019-08-20:
There was a recent [proposal](https://lists.riscv.org/g/tech-privileged/message/640) to add a bit to fcsr to indicate support for half-precision FP.  An [argument](https://lists.riscv.org/g/tech-privileged/message/642) was made that this was undesirable partially due to headaches for classical virtualization.

Is it any more desirable to expose WARL bits to U mode in fcsr without the ability to trap the fcsr read access (other than the big hammer of trapping all FP operations)?  The term WARL is only in the privileged architecture and there are no WARL bits in the unprivileged ISA.

aswaterman, 2019-08-20:
The virtualization argument was specifically about the register behaving
differently in different privilege modes, which does not apply here.

Any proposal that adds new fields to FCSR really must be WARL, because
existing implementations will read those fields as zero.

pdonahue-ventana, 2019-08-20:
I just want to bring to your attention that software cannot emulate an implementation that supports non-IEEE half-precision without trapping all FP instructions of all sizes (even if single and double are natively supported or software doesn't actually use non-IEEE half-precision), solely because there is no mechanism to trap accesses to fcsr without trapping accesses to all FP.

I see that ARMv8 has the same limitation so perhaps nobody cares about this case.

aswaterman, 2019-08-20:
Yep.  That concern should be weighed when making this decision.

eak, 2019-08-20:
This could be viewed as a reason to get it into the ISA as soon as possible, so implementations can add the bit(s) and trap if it is set if they don't implement bfloat16. It is another reason to make it a 2-bit field initially (again so they can trap if a non-zero value is there).

aswaterman, 2019-08-20:
Indeed.

eak, 2019-08-20:
Do we need to move this to the tech mailing list?

aswaterman, 2019-08-20:
It's being discussed in tech-privileged

nihui, 2021-06-11:
Hi, what is the status now ?
Is there any way for non-privileged code to discover if the processor supports risc-v zfh extension ?
We need a runtime detector function for dispatching proper optimized code path based on processor capability.
Thanks.

gfavor, 2021-06-11:
A new "unified low-level RISC-V discovery method" is in the works.  This will support discovery, for all arch extensions, of the support for the extension as well as the support for any options within the extension and supported values for any key parameters within the extension.

Note that this is the "low-level" discovery method that, for example, M-mode software would use to populate info into high-level discovery structures such as Device Tree.  The "runtime detector function" obviously is a function of the system environment (e.g Linux, RTOS, bare-metal, ...) and how the toolchain and loader and user discovery are handle all this.

aswaterman, 2024-03-29:
I think @gfavor’s comment from a few years ago is conclusive.

closed 2024-03-29 by aswaterman

nihui, 2024-03-30:
Hello everyone, if you want to implement detection on an old version of the kernel, the ruapu project is a good reference. It handles sigill and completes the detection of the extended instruction set.

https://github.com/nihui/ruapu