riscv/riscv-isa-manual #415: Canonical order for Z* extensions

kito-cheng, 2019-07-23:
Table  27.1  has listed `Zicsr`, `Zifencei`, `Zam`, `Ztso` and `Zx*`, and said it's also canonical order, does it mean `IMADZifencei_C` is correct and `IMADCZifencei` is incorrect order ? if later one is correct, table 27.1 might need to put all `Z*` in later row?

And further order question: 
- Order between `Zxm*` and `Z*`? I assume in general `Zxm*` appear after `Z*` because alphabet order, but in case we might have `Zz*` extension in future?
- Order between `Z*` and others (`S`, `H` and `X`)

aswaterman, 2019-07-23:
This is a bug in the table, or perhaps a bug in the caption.  `IMADCZifencei` is valid; `IMADZifencei_C` is not.

I think we should clearly state that Z extensions follow all other standard extensions (i.e. after all other letters except X), and X extensions follow all standard extensions (i.e. after all other letters).

Let's keep the rules simple and have Zxm ordered alphabetically.  It's OK if Zz extensions come after Zxm extensions.

kito-cheng, 2019-07-24:
Got it, there is some people begin implement `-march` parser for `Z*` on B taskgroup[1], so I think it's time to implement that on `binutils`/`gcc`.

[1]https://github.com/riscv/riscv-bitmanip/issues/38

aswaterman, 2019-07-24:
Agreed.

I've fixed the table, and I fixed the text to indicate X-extensions must come last.  This is not a complete fix, but it will help avoid some confusion.  Can you check this commit and see if you agree with it? e87173489d7cf3818aecaf397b7e54f787cc2d31

kito-cheng, 2019-07-24:
Hmm, order between `Zicsr`, `Zifencei`, `Zam` and `Ztso` should in alphabet order in table 27.1?

`Zam`, `Zicsr`, `Zifencei`, `Ztso`

aswaterman, 2019-07-24:
I forgot that Z extensions are ordered the same way as base extensions:
that is, Zi Zm Za Zf Zd Zc...

So yes, this issue is still not resolved.

marceg, 2019-07-24:
Alphabetical is simple and predictable, allows coding now for any future extension. One option is to be a bit disruptive and apply this to all including base extensions. Short of that, what about only specifying ordering of currently commonly used and ratified base extensions (IEMAFDC, E is not common but takes the place of I) and leaving everything else alphabetical?  Including among Z*, and possibly X* ordered last.

The simplicity in understanding and coding seems worth a slight bit of disruption.

aswaterman, 2019-07-24:
There are already many upstream tools that enforce the current IMAFD...
ordering, so it would be extremely disruptive to change that.

Since nothing depends on the current Z ordering AFAIK (unless @asb knows
otherwise?), now is a good time to make sure we are getting that part
right. Making it fully alphabetized seems like a good simplification to me.

kasanovic, 2019-07-24:
For conciseness, the letters are context-sensitive and the canonical
ordering is meant to avoid having to contemplate all partial orders.

E.g., RV32IDC - what is in C, depends on whether D is present, whereas
D does not depend on C.  Or more simply, RV32ID and RV64ID -what D
means varies depending on prefix.

My original thought was that Z followed this model, and separately we
have fully disambiguated strings in a feature map (similar to what
Nikhil has been working on in context of compliance).  The fully
disambiguated strings would look something like:

 "rv32i-f-fadds"

Krste

| There are already many upstream tools that enforce the current IMAFD...
| ordering, so it would be extremely disruptive to change that.

| Since nothing depends on the current Z ordering AFAIK (unless @asb knows
| otherwise?), now is a good time to make sure we are getting that part
| right. Making it fully alphabetized seems like a good simplification to me.

| On Wed, Jul 24, 2019 at 6:47 AM Marc Gauthier <notifications@github.com>
| wrote:

|| Alphabetical is simple and predictable, allows coding now for any future
|| extension. One option is to be a bit disruptive and apply this to all
|| including base extensions. Short of that, what about only specifying
|| ordering of currently commonly used and ratified base extensions (IEMAFDC,
|| E is not common but takes the place of I) and leaving everything else
|| alphabetical? Including among Z*, and possibly X* ordered last.
|| 
|| The simplicity in understanding and coding seems worth a slight bit of
|| disruption.
|| 
|| —
||
|| 
|| 
|| 
| <https://github.com/riscv/riscv-isa-manual/issues/415?email_source=notifications&email_token=AAH3XQWQVV446OTMFBIWZZDQBBMQNA5CNFSM4IGEKLTKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2WMNOY#issuecomment-514639547>
| ;,
|| .
|| 

| —

kito-cheng, 2021-07-06:
I believe this issue is clarified and both implemented on GNU and LLVM toolchain.

closed 2021-07-06 by kito-cheng