riscv/riscv-isa-manual #419: `fssr` and `frsr` instructions aren't mentioned for F extensions

laanwj, 2019-07-29:
Just a documentation issue I think: The `fssr` and `frsr` instructions (which set and read the `fcsr`, respectively) aren't mentioned in the specification for the F extension. The names don't appear in the spec at all.

aswaterman, 2019-07-29:
They’re pseudoinstructions that map to CSR-access instructions. (And they
may well be missing from the documentation.)

laanwj, 2019-07-29:
Oh.

I'm confused now, this came up in https://github.com/rust-embedded/riscv-rt/issues/29, there was the assumption that these were separate instructions, but if they map to the same thing then it's probably not necessary to mention them.

closed 2019-07-29 by laanwj

Mel-Chen, 2019-07-30:
Hi! I also noticed this issue recently. Are `frsr` and `fssr` instructions obsolete, or renamed to `frcsr` and `fscsr` respectively? Thanks.

aswaterman, 2019-07-30:
Correct.  But the obsolete aliases will be supported in perpetuity, since
important software (like glibc) relies upon them.

laanwj, 2019-07-30:
Good to know!
RISC-V `objdump` as of version 2.32 still generates those aliases on disassembly.

Mel-Chen, 2019-07-30:
Don't worried. I am preparing binutils patches to update `objdump`. :P