riscv/riscv-isa-manual #424: Add CSRs to the hypervisor extension for more trap values

jhauser-us, 2019-07-31:
This issue is a variation on #394.

There are reasons to suspect that hypervisor performance could substantially benefit if trap handlers received more information about page faults caused by guest-physical address translation (second-level translation).

To test this hypothesis, I propose adding two hypervisor CSRs, `htval` and `htinst`, plus two machine-level CSRs, `mtval2` and `mtinst`.  In the names, "tinst" is short for "trap instruction".  Although the hypervisor extension would require all four CSRs to exist, a minimal implementation would be free to hardwire all of them to zeros.  Like `mtval` and `stval`, `htval` and `htinst` must be written automatically on any trap into HS mode, and `mtval2` and `mtinst` must be written on any trap into M mode.

When a trap is taken into HS mode due to guest-physical address translation (`hstatus`.STL = 1), `htval` is automatically written either with zero or with the guest physical address that faulted, shifted right by 2 bits.  For any other trap into HS mode, `htval` must be written with zero.

When a trap is taken into HS mode, `htinst` is automatically written with either
 - zero;
 - the trapping instruction in uncompressed form; or
 - a special value.

Except when a special value is required (described later), the value assigned to `htinst` may always be zero, indicating that the hardware isn't providing the trapping instruction for this particular trap.  If the value assigned to `htinst` is neither zero nor a special value, it is determined as follows:
 - For an instruction that is not a compressed instruction (32-bit size or larger), the value is the complete instruction if its size does not exceed HSXLEN, or the instruction's least-significant HSXLEN bits.
 - For a compressed instruction (16-bit size), the value is not the actual instruction but instead is the non-compressed (32-bit) instruction into which the compressed instruction expands, except with bit 1 set to 0.

Note that, when an instruction is written to `htinst`, bit 0 is always 1, and bit 1 indicates whether the original instruction was compressed (16 bits).

For an instruction address exception or instruction access fault or page fault, `htinst` must be set to zero, as there's no instruction available.  Also, if a compressed instruction causes an illegal instruction trap, again `htinst` must be zero, because the instruction's expansion isn't defined.  (But the compressed instruction may be in `stval`.)

Special values for `htinst` are nonzero values having bit 0 = 0.  (That is, they are nonzero *even* values, whereas instructions written to `htinst` are always *odd*.)  If a trap into HS mode is a page fault or access fault due to guest-physical address translation, and `htval` is given a nonzero value (the faulting guest physical address), and furthermore if the guest memory access was for VS-level address translation, then `htinst` must be written with one of these special values.

       2 = an implicit read for address translation
    0x22 = an implicit write for address translation

Note that, if an implementation uses `htval` to supply the faulting guest physical address to a trap handler, it cannot hardwire `htinst` entirely to zero.  However, it may choose to support only values 0 and 2, and possibly 0x22.  (Value 0x22 is needed only if the hardware may automatically update bits A and D in guest-physical page tables, else it won't arise.)  On the other hand, if `htinst` bit 0 is writable, all of the least-significant 32 bits of `htinst` must be writable.

With these new CSRs, a trap handler for guest-physical page faults will have this information available to it:
 - in `stval`, the original virtual address;
 - in `htval`, if nonzero, the faulting guest physical address (shifted right by 2 bits); and
 - in `htinst`, either a special value indicating that the fault occurred due to a read or write by VS-level address translation, or, if nonzero, the uncompressed faulting instruction.

When maximally implemented, this should both speed up the handling of guest-physical page faults and facilitate the emulation of I/O devices for a guest virtual machine.

Machine-level CSRs `mtval2` and `mtinst` are of course defined identically to `htval` and `htinst`, except for M mode.

jhauser-us, 2019-08-01:
I wrote:
> (Value 0x22 is needed only if the hardware may automatically update bits A and D in guest-physical page tables, else it won't arise.)

Instead of "guest-physical page tables", I should've said "VS-level page tables".  Sorry for any confusion.

fintelia, 2019-08-01:
I quite like this proposal.

One tweak I'd like to see is to have the "special values" encoded in `htval`. Under all current paging modes when `htval` is nonzero the contents of `htval[7:0]` will be equal to `stval[9:2]`. I'm assuming this redundancy is intended to support alternative translations modes like segmentation which would have alignments far less than 4KB but if there were restricted to at least 8 bytes there is a trick to reclaim the redundant bits in the CSR: have htval[HSXLEN-1]=1 mean that the guest physical address has been shifted right by *10* and reserve htval[HSXLEN-1]=0 to mean a smaller shift.

I'm also curious what the logic was for the value 0x22 in particular? Presumably it matches the bit pattern from somewhere else or otherwise 0x4 or 0x6 would have been chosen?

jhauser-us, 2019-08-01:
Jonathan,

You've made a good observation about the partial redundancy between nonzero `htval` and `stval`, although I believe it should be `htval[9:0]` equals `stval[11:2]`, no?

For myself, this redundancy doesn't bother me, since, as you say, we or somebody might find it desirable to support "alternative translation modes like segmentation which would have alignments far less than 4 KB".  But I'm open to hearing other opinions on the matter, if people feel we should avoid or exploit the redundancy.

If I understood you correctly, instead of the hardware writing special values to `htinst`, you suggest reducing the redundant bits in `htval` and reusing some of them to encode the same information.  However, you don't explain why you believe this change would be an improvement.  Do you see a problem with the proposed scheme for special values in `htinst`?

Yes, the 0x22 is a bit mysterious.  (There's a pun there, if you want it.)  The value is explained by the patterns you find in bits 6:0 of uncompressed load, store, and AMO instructions, as they would appear in `htinst`:

    00000x1   LOAD instruction
    00001x1   LOAD-FP instruction
    01000x1   STORE instruction
    01001x1   STORE-FP instruction
    0101111   AMO instruction

    0000010 (2)     implicit read for VS-level address translation
    0100010 (0x22)  implicit write for VS-level address translation

The numbers 2 and 0x22 I chose aren't sacrosanct, if anyone wants to argue for a different pair.

avpatel, 2019-08-01:
Overall, I am fine with the proposal.

The proposed change in stval CSR interpretation means we have to change SW (Xvisor and KVM) on how we determine faulting guest physical address.

jhauser-us, 2019-08-01:
Anup Patel wrote:
> The proposed change in stval CSR interpretation means we have to change SW (Xvisor and KVM) on how we determine faulting guest physical address.

I'm afraid I didn't understand this.  What proposed change to `stval` interpretation do you mean?  The proposal here does not intend to change anything written by the hardware to `stval`.

avpatel, 2019-08-01:
For stage2 load/store page faults, STVAL CSR contains guest physical address as-per hypervisor v0.4 spec. This proposal states that STVAL CSR will now contain original virtual address which is not backward compatible with hypervisor v0.4 spec.

jhauser-us, 2019-08-01:
Anup Patel wrote:
> For stage2 load/store page faults, STVAL CSR contains guest physical address as-per hypervisor v0.4 spec.

Evidently, the current hypervisor draft is not as clear as it needs to be on this point.  Section 4.1.9, "Supervisor Trap Value (`stval`) Register", says
> When a ... page-fault exception occurs, `stval` is written with the faulting virtual address.

When two-level address translation is in effect, it was always intended that this still be the *original* guest virtual address, never the guest physical address.  But I can see that "faulting virtual address" is ambiguous in this situation, so the hypervisor chapter needs to be clarified.  (Perhaps this is actually stated somewhere, but I couldn't find it easily.)

So, yes, you are right, the hypervisor implementations will need to change, but not because of this proposal.  Rather, there's been an unfortunate misunderstanding about what `stval` was supposed to contain all along.

(Consider that, for RV32, a guest physical address can be 34 bits.  It should be clear that not all 34 bits can be stored in 32-bit `stval`, so which bits was it assumed were kept?  I suppose bits 31:0, dropping 33:32?)

avpatel, 2019-08-01:
> So, yes, you are right, the hypervisor implementations will need to change, but not because of this proposal. Rather, there's been an unfortunate misunderstanding about what `stval` was supposed to [...]

I support your proposal. It is much cleaner. I just wanted highlight the SW implications.

fintelia, 2019-08-01:
> If I understood you correctly, instead of the hardware writing special values to htinst, you suggest reducing the redundant bits in htval and reusing some of them to encode the same information. [...]

From a functionality perspective I think the htinst scheme is fine, but it feels kind of inelegant to me.  As currently written, the htinst register holds the current instruction or zero for every trap into HS-mode from VS-mode, VU-mode, and normal U-mode except for one very specific hypervisor edge case where it instead *must* have a special value (there could be other edge cases in the future, but for backwards compatibility the extension would say that it *may* hold a special value). The description of when htinst can be hardwired even entangles htinst with htval. If it weren't for these small details, mtinst seems like a very natural standalone extension that would be useful independent of hypervisors.

It also occurs to me that the redundant bits in mtval2 could act almost like 10 additional free bits in mstatus (20 if you count htval also). For instance, instead of having mstatus.MTL there could just be a mtval2.MTL bit. Phrased this way it feels hard to justify surrendering the bits so easily.

> Also, if a compressed instruction causes an illegal instruction trap, again htinst must be zero, because the instruction's expansion isn't defined. (But the compressed instruction may be in stval.)

There is some subtly here. `c.unimp` generates an illegal instruction trap even though its expansion is well defined. The same is true of compressed floating point instructions while the relevant extensions are disabled. I think "must" -> "may" resolves these.

jhauser-us, 2019-08-02:
In another forum, a case has been made (or perhaps made again) that asking the hardware to supply the complete trapping instruction in `mtinst`/`htinst`, even in uncompressed form, is overkill for our needs.  After I find time to review and rethink this, there's a good chance I'll withdraw `mtinst`/`htinst` and back a different way to supply a trap handler with the information it really needs.

jhauser-us, 2019-08-03:
This issue has been replaced by version 2, #431.

closed 2019-08-03 by jhauser-us