riscv/riscv-isa-manual #425: Accessing SIP.SSIP, SIP.STIP, and SIP.SEIP bits from HS-mode and VS-mode

avpatel, 2019-08-01:
We need some clarity on accessing S-mode interrupt bits in SIP CSR from HS-mode and VS-modes.

As-per my understanding, all S-mode interrupt bits in SIP CSR read-only for both VS-mode and HS-mode. This rationale is based on MIP CSR documentation in RISC-V spec which says:
"The mip register is an MXLEN-bit read/write register containing information on pending interrupts, while mie is the corresponding MXLEN-bit read/write register containing interrupt enable bits. Only the bits corresponding to lower-privilege software interrupts (USIP, SSIP), timer interrupts (UTIP, STIP), and external interrupts (UEIP, SEIP) in mip are writable through this CSR address; the remaining bits are read-only."
(Note: SIP CSR documentation in RISC-V spec does not clearly tell which bits are accessible in S-mode)

The HS-mode can certainly write to S-mode interrupt bits in VSIP CSR which will be readable from VS-mode as SIP CSR.

Can VS-mode update S-mode interrupt bits in SIP CSR (i.e. VSIP CSR) ???

Regards,
Anup

bonzini, 2019-08-02:
>  (Note: SIP CSR documentation in RISC-V spec does not clearly tell which bits are accessible in S-mode)

The spec says that "All bits besides SSIP in the sip register are read-only", and "A supervisor-level software interrupt is triggered on the current hart by writing 1 to its supervisor software interrupt-pending (SSIP) bit in the sip register."

Certainly it does not qualify who can write 1 to SSIP, but I don't think you can compare SSIP to MSIP since MSIP is read-only while SSIP is not. Therefore I'm not sure why sip.SSIP should be writable only by M-mode (or vsip.SSIP should be writable only from HS-mode0.

aswaterman, 2019-08-30:
cc @jhauser-us

avpatel, 2019-10-19:
The SIP.SSIP bit is indeed writable from both HS-mode and VS-mode.

Closing this issue now.

Regards,
Anup

closed 2019-10-19 by avpatel