riscv/riscv-isa-manual #426: Consider requiring that scause bits 4:0 all be implemented

jhauser-us, 2019-08-01:
The privileged spec's requirements for `scause` effectively force any practical implementation to have bits 3:0 be fully implemented as four writable flip-flops.  It has been conjectured that most if not all existing implementations of `scause` actually have five writable flip-flops for bits 4:0.

In order to provide a set of known usable values for `scause` beyond solely the 17 exception codes defined at supervisor level (3 for interrupts, 14 for synchronous exceptions), I propose that the spec be amended to require that `scause` be capable of holding all values in the range 0 through 31, plus all values in the range 2<SUP>(SXLEN - 1)</SUP> through 2<SUP>(SXLEN - 1)</SUP> + 31.

The ability to store in `scause` values other than the 17 exception codes currently defined can be a prerequisite for machine-level software to be able to emulate other exception causes for S mode.  In particular, the ability to store values 20, 21, and 23 in `scause` may soon be a prerequisite for emulating hypervisor support in a machine having only M/S/U modes.

The proposed new requirement would apply only to `scause`, not to `mcause`.

aswaterman, 2019-08-01:
I'm all for this.

jhauser-us, 2019-08-01:
Not that it's important, but I miscounted.  There are 16 exception codes currently defined at supervisor level:  3 for interrupts and 13 for synchronous exceptions.

avpatel, 2019-08-02:
Looks good to me.

commit 07a780c6 (2019-08-27): scause must be able to hold the values 0-31

Resolves #426
closed 2019-08-27 by aswaterman