riscv/riscv-isa-manual #427: Require that mstatus.MPRV become 0 on transitions to lesser privilege modes

jhauser-us, 2019-08-01:
The hypervisor extension defines an SPRV field in `hstatus` that performs for HS mode a function similar to the `mstatus`.MPRV available to M mode.  In order both to greatly simplify the machine states associated with `hstatus`.SPRV and to narrow a possible security risk, when any MRET or SRET changes the operating mode to one less privileged than HS mode, SPRV is required to become 0.

For consistency and to obtain the same advantages, I propose that `mstatus`.MPRV be required to become 0 when an MRET changes the mode to one less privileged than M mode.

Although this is technically a change to hardware behavior, respectable software is already expected not to depend on MPRV = 1 outside of M mode.  Hence, no existing respectable software should be impacted.

jscheid-ventana, 2019-08-02:
How is this different in effect than specifying that *PRV is treated as zero when operating at a lower privilege mode?

Perhaps *PRV should be cleared when moving to the associated mode from lower?

aswaterman, 2019-08-02:
I had forgotten that, per the debug spec, MPRV is ignored when in D-mode.  So if we were to make a change to this feature here, it might be more consistent to ignore MPRV when the mode is not M, rather than resetting MPRV when leaving M for a less-privileged mode.  But I guess either solution works.

jscheid-ventana, 2019-08-02:
If the choice is to ignore, then should there be a HW requirement to clear *PRV when entering the mode from below?  This would avoid concern about a certain class of SW bug.

aswaterman, 2019-08-02:
If avoiding that class of bug is part of the concern, then JohnH's original proposal (zero the bit when leaving M-mode for below) seems simpler, since it's only one mechanism (zero the bit on exit) vs. two mechanisms (ignore; also zero the bit on entry).

jscheid-ventana, 2019-08-03:
I don't have an opinion on avoiding the SW bug I mentioned above.

I think ignoring the state is preferable to changing the state.

Is it made clear whether xPRV is intended to only apply when in mode 'x', or if it's additive/cascading (M uses S which uses U)?

marceg, 2019-08-03:
If trap entry code takes a trap while saving state because MPRV=1, it takes a trap again, likely hitting the same issue again, in an infinite loop.  Unless trap entry code anticipates that somehow.  Clearing MPRV avoids this, if somehow M-code can be tricked into exiting with it set (unlikely as that may be).

As defined, MPRV and SPRV cascade:  "When MPRV=1, load and store memory addresses are translated and protected as though the current privilege mode were set to MPP." Lower privilege modes act upon SPRV.  Drawing out the combinations is convoluted as you might guess (or more).

aswaterman, 2019-11-20:
Closed by 569d07195a8495460f04592d8455153f730a0f54

closed 2019-11-20 by aswaterman