riscv/riscv-isa-manual #428: Replace hypervisor mstatus.MTL, hstatus.STL with separate exception codes

jhauser-us, 2019-08-02:
The current hypervisor draft defines a dedicated bit in `mstatus`, called MTL, to indicate to M mode whether a page fault or access fault was due to second-level address translation or due to some other cause.  For traps delivered to HS mode, `hstatus` bit STL provides the same information.

It is proposed to eliminate bits MTL and STL and instead define new "guest-page fault" exceptions with these codes:

    20 = instruction guest-page fault
    21 = load guest-page fault
    23 = store/AMO guest-page fault

Page faults due to second-level address translation would be indicated by one of these new "guest-page fault" codes in `mcause` or `scause`.  Other page faults would continue to use existing codes 12, 13, and 15.

This issue is related to #426.

With this change, it would no longer be possible to distinguish access faults due to second-level address translation from other access faults.  Hypervisors that misconfigured guest physical page tables would cause a guest virtual machine to see access faults for what should have been valid memory accesses.  However, this might be no worse than the consequences of many other potential hypervisor mistakes.

The principal (and perhaps only significant) advantage of this change would be to improve ISA consistency by using an existing mechanism (different codes in `*cause`) to distinguish second-level page faults from other traps.

avpatel, 2019-08-02:
Looks good to me.

yyshen, 2019-08-07:
Could you elaborate the sentence below a bit?
 
" Hypervisors that misconfigured guest physical page tables would cause a guest virtual machine to see access faults for what should have been valid memory accesses."

I wonder why a guest VM would see an access fault caused by mis-configured second-stage translation. Should not the second-stage translation faults be handled by a hypervisor or VMM? My understand is the guest VM should not be able to observe stage-2 faults. Maybe I misunderstood something?

Thank you!

marceg, 2019-08-07:
I'm guessing it means the hypervisor setting up a PTE to a physical 
address which it does not have access to per PMP entries setup by M-mode.

It seemed unusual to delegate access faults.  Usually M-mode gets a 
first look, then delegates in software, to the next mode, and so on.  So 
I suspect the hypervisor will see it before the guest does.

-M

yyshen wrote on 2019-08-07 17:51:

jhauser-us, 2019-08-14:
I wrote:
> Hypervisors that misconfigured guest physical page tables would cause a guest virtual machine to see access faults for what should have been valid memory accesses.

That's assuming all access faults are delegated straight through to VS mode, which need not be the case.  Marc Gauthier may be right:
>It seemed unusual to delegate access faults.  Usually M-mode gets a first look, then delegates in software, to the next mode, and so on.  So I suspect the hypervisor will see it before the guest does.

avpatel, 2019-08-31:
Any updates on this ?
Can we have a PR for this ?

jhauser-us, 2019-11-07:
Adopted with commit 899457caf7a9cdcb59a7d57421d249512b00de94.

closed 2019-11-07 by jhauser-us