riscv/riscv-isa-manual #429: New HGIP CSR for injecting interrupts to VS-mode

avpatel, 2019-08-02:
The current method of interrupt injection in VS-mode is to directly
update VSIP CSR from HS-mode.

This approach works fine if HS-mode software is the only entity
injecting VS-mode interrupts but it becomes problematic if we have an
external entity (such PLICv2 HW with virtualization support) capable
of injecting VS-mode interrupts.

I propose we have a new HGIP CSR (Hypervisor guest interrupt pending)
for HS-mode software. The RISC-V CPU implementations can also have HW
VS-mode interrupt lines. The VSIP CSR will be an ORing of HGIP CSR and
HW VS-mode virtual interrupt lines. This way HS-mode and/or external
entity (such as PLICv2) can inject virtual interrupts to VS-mode.

With new HGIP CSR in-place, SIP CSR will be read-only for VS-mode and
hypersiors in HS-mode will save/restore HGIP CSR instead of VSIP CSR.

bonzini, 2019-08-02:
> With new HGIP CSR in-place, SIP CSR will be read-only for VS-mode and
hypervisors in HS-mode will save/restore HGIP CSR instead of VSIP CSR.

Doesn't this contradict the current spec, which says that sip.SSIP is *not* read-only for S-mode?

avpatel, 2019-08-02:
> > With new HGIP CSR in-place, SIP CSR will be read-only for VS-mode and
> > hypervisors in HS-mode will save/restore HGIP CSR instead of VSIP CSR.
> 
> Doesn't this contradict the current spec, which says that sip.SSIP is _not_ read-only for S-mode?

Ahh yes, SIP CSR cannot be entirely read-only. It also have user interrupt bits. Please ignore the last statement.

bonzini, 2019-09-23:
Given the last comment from @avpatel (as well as the experience from porting KVM to RISC-V), this doesn't seem to be required. Can it be closed?

avpatel, 2019-09-23:
Yes, I thought about this more.

The HGIP CSR won't be required for supporting external entity (such as PLICv2) inject VS-mode interrupts.

Let's say at any point in time the register memory backing VSIP CSR has value X.

Now, the way to support HW VS-mode interrupt lines can be as follow:
1. Reads to VSIP CSR will be ORing of HW VS-mode interrupt lines and X
2. Writes to VSIP CSR will only update X (internal register memory)

Classic example of above is the HCR_EL2.VF and HCR_EL2.VI bits defined in ARM64 architecture.

I will wait 1 more day for comments and close it myself.

closed 2019-09-25 by avpatel