riscv/riscv-isa-manual #435: Priority of Misaligned, Page and Access Faults

vmurali, 2019-08-09:
Table 3.7 lists that for Loads/Stores/AMOs, Misaligned faults should be raised before Page and Access Fault. If the processor refuses to do any misaligned accesses, then this should be fine. But, if the processor consults the memory/IO region to decide whether a misaligned access is allowed, then we can obtain the address only after address translation, only after which we can determine if it is misaligned. Perhaps a fix would be to make the priority implementation dependent?

aswaterman, 2019-08-09:
I do think we should relax this requirement for v1.12, and leave it up to the implementation whether page faults or misaligned faults have priority.  (I don't want to just change the priority, because it creates an unnecessary backwards incompatibility with v1.11 hardware.)

jscheid-ventana, 2019-08-12:
The current specification already seems to accommodate misaligned-as-access faults.  So I'm not sure what compels any change.

aswaterman, 2019-08-12:
The issue is one of exception priority.  The exception-priority table indicates that misaligned exceptions take precedence over page faults, which would seem to be a contradiction if address translation needs to be carried out in order to know whether a misaligned fault should be raised.

marceg, 2019-08-12:
FWIW, there's a precedent for having two different priorities of misaligned exceptions in another architecture where hardware support for misaligned accesses is optional.  When not present, all misaligned accesses are trapped at a relatively high priority.  When support is present, some devices in physical space might not support it (or not support accesses crossing two memories, etc), in which case the misaligned exception has lower priority.  Priority is fixed for a given configuration.

aswaterman, 2019-08-13:
Thanks @marceg.  When @vmurali and I discussed this earlier, we also concurred that, while it seems benign to leave this implementation-defined, it should be fixed for a given implementation.

jscheid-ventana, 2019-08-13:
So disallowing the misaligned-as-access cases, or still allowing those, but requiring a consistent ordering between, say, misaligned faults and page faults?

vmurali, 2019-08-13:
I believe this is orthogonal to misaligned-as-access exceptions. If misaligned accesses were trapped as misaligned exceptions, we need to determine priority between misaligned exceptions and page faults. For a particular implementation, this priority is fixed, but different implementations can have different priorities between the two.

aswaterman, 2019-08-13:
The concrete proposal is to make the priority of page faults vs. misaligned implementation-defined, and to change nothing else.

marceg, 2019-08-13:
I can only think of two useful prioritizations, rather than completely implementation-defined.

jscheid-ventana, 2019-08-13:
I'm trying to understand the intent of enabling designs to prioritize page faults over misaligned faults.  From Murali's first comment:
`if the processor consults the memory/IO region to decide whether a misaligned access is allowed`
In that scenario, isn't that by definition a PMA-based access fault?

Perhaps there's some philosophical basis that I'm not getting.

marceg, 2019-08-13:
Good question, tripped on it some time ago.  Hiding in the atomicity PMAs section is this:

> Implementations may raise access exceptions instead of address-misaligned exceptions for some
> misaligned accesses, indicating the instruction should not be emulated by a trap handler.

So the choice between the two appears to depend on whether the access can succeed by emulation that for example does the access one byte at a time.

marceg, 2019-08-13:
Granted, the assumptions in that hidden statement contradict current exception priorities.

aswaterman, 2019-08-13:
@marceg's initial post accurately summarized the in-person conversation I had with @vmurali.  Let's not hypothesize further without additional input from the OP.

vmurali, 2019-08-13:
I agree with @marceg 's post and later discussion. The concrete proposal is to keep the priority between Page Fault and Misaligned Accesses implementation dependent, while keeping the rest of the priorities as is. The implementation dependent priority is determined during configuration, and cannot be altered dynamically.

(As @marceg point out, the difference between misaligned fault and access fault is that the misaligned trap is supposed to be handled by the trap handler, whereas access fault is a catastrophic failure. So, scratchpad memories that don't support misaligned accesses can still emulate such accesses with multiple aligned memory requests via a trap handler for the misaligned fault; if it raised an access fault instead, the trap handler would simply give up. The current priority scheme is good for processors that always emulate misaligned accesses using the trap handler)

dkruckemyer-ventana, 2019-08-27:
> @vmurali <https://github.com/vmurali> can you propose edits to the spec?

Though I understand the motivation of classifying page fault exceptions
above misaligned exceptions (since many implementations need to clear the
former before detecting the latter), does it matter practically? In other
words, for these two exceptions (or including access exceptions), would
software be written differently knowing the priority order? If not, it
seems that it should be OK for hardware to pick one arbitrarily based on
whichever was detected first (especially if software can't tell or act on
the difference), and the architecture can just specify that they have the
"same" priority. (I guess this is basically the same as
"implementation-defined priority" in a sense....)

Cheers,
David

aswaterman, 2020-06-11:
I've attempted to resolve this issue in 9ff515cd6695ac392e5ca32b73a135aa197e2778

closed 2020-06-11 by aswaterman