riscv/riscv-isa-manual #436: Bitmanip clash with R-type fmt & clash(?) with OP128 (page 42 of ISA manual)

xphung, 2019-08-13:
**Q1: Should two source reg + 1 dest reg (R-type like) operations go into OP or OP-IM?**

Below is the v0.90 BitManip proposed encoding of FSRI:

`
rs2 | func2 | imm6 | rs1 | func3 | rd | OP-IM
`

where func3 = 0b101, func2 = 0b10, OP-IM  = 0b0011011

Note:
(1) FSRI is effectively an "R-type" instruction (two source regs, one dest reg) but uses it's own format which is different to the standard R-type instruction format
(2) FSRI is the only instruction to use this format, so this new format is a "one off" for a single instruction
(3) FSRI is placed into the OP-IM major opcode, which to date only contains I-type (single source reg) instruction formats (again, another "one-off")
(4) OP-IM is very constrained in terms of unused opcode space, as non-shift instructions already fully use all the 12 immediate bits, and the RV64 shift instructions leave only 6 bits for a func6 opcode (vs 7 bits in func7 of the OP major opcode).

**Q2: Should the possibility of binary compatibility between RV128 and RV64 be preserved by all standard extensions?**

Page 42 of the latest draft of RISC V ISA manual hints binary compatibility may be a future goal:

**`
To improve compatibility with RV64, in a reverse of how RV32 to RV64 was handled, we might change the decoding around to rename RV64I ADD as a 64-bit ADDD, and add a 128-bit ADDQ in what was previously the OP-64 major opcode (now renamed the OP-128 major opcode).
`**

I note the use of past tense above ("now renamed the OP-128 major opcode")...
Does this mean the choice of OP-128 (over OP-64) is now highly likely?  (I am in favour if so).

Even if it is premature to make a decision (given RV128 might be some time away), should we be making active decisions now (eg. BitManip) to define instructions (eg. FSRI) that are excluded from RV128 (which the current 0.90 spec states as a policy)?  This would mean RV128 can't have binary compatibility with RV64 (as it won't be a superset of RV64).

[Or if FSRI is then retrofitted into a future RV128 to be binary compatible with RV64, then it would come at considerable cost to OP-IM encoding space, ie: using up 5 of the available 6 func bits in the shift-like I-type OP-IM instructions].

I have already raised the above two issues with BitManip's editor but he has decided it was a broader RISC V opcode question that was outside scope of BitManip.
==> This seems fair enough, I am in favour of having FSRI in BitManip, I just think it should be encoded differently
ie: in OP rather than OP-IM, given that it's essentially a 2 source register R-type instruction and should therefore use the standard R-type format and not create a new single instruction format on it's own.  This will allow it to be available in both RV64 and RV128 whilst only using 6 out of the 10 bits (ie: 64/1024 = 6%) of opcode space in OP's func7+func3 - see (https://github.com/riscv/riscv-bitmanip/issues/42) for how this encoding could work.

aswaterman, 2019-08-13:
I think @cliffordwolf will have to help with this discussion.  I don't understand why rs2 isn't in the usual place, so it's hard to comment on the rest of question 1.

I had imagined RV128 machines would run RV64 code with a mode switch (UXL/SXL/MXL), so it is possible there will be instructions that are legal in RV64 mode but illegal in RV128 mode.  In my mind, the main reason for the change of encoding for ADDD in RV128 is to make supporting both modes cheaper; the goal wasn't to run RV64 code in RV128 mode.  Would you agree, @kasanovic?

xphung, 2019-08-14:
> I think @cliffordwolf will have to help with this discussion. I don't understand why rs2 isn't in the usual place, so it's hard to comment on the rest of question 1.
> 

I hope @cliffordwolf doesn't mind me quoting his previous reply when I asked him about rs2 being moved (note there is a "cosmetic" renaming of rs2 into rs3 in his terminology):

>> No, the format doesn't move rs2 to another position. Like with all immediate instructions, rs2 is replaced by the immediate. The new field is rs3 and it is in the usual rs3 position.

My translation/interpretation (apologies in advance if it is not a correct interpretation) is he wants to use a R4-like instruction format for FSRI because he also happens to use the R4 instruction format for FSR and FSL.  However, FSR and FSL are true R4 instructions as they have three input registers - this is not the case for FSRI, which has two register inputs and an immediate 6 bit shift amount.  So he made (IMHO an arbitrary & wrong) choice to modify R4 by dropping rs2 and putting in an immediate field in it's place.  (My opinion is that the good old R-type format is best for FSRI, no need to invent a new immediate version of R4).

xphung, 2019-08-14:
> In my mind, the main reason for the change of encoding for ADDD in RV128 is to make supporting both modes cheaper; the goal wasn't to run RV64 code in RV128 mode. Would you agree, @kasanovic?

Can I clarify if the RV shift & multiply will also follow encoding of ADDD/ADDQ?  (ie: double word operations in OP and quad word operations in OP-128)

To "sell" the idea of keeping all double word operations in OP, some possible benefits are:
(1) Potential of binary compatibility between RV64 and RV128, but also
(2) Only need to reserve 6 bits for shift fields in OP-IM, and avoid the "wastage" of bits like what occurred when 5 bit shift fields (RV32) went up to 6 bit (RV64)... ie: retains use of all 6 func code bits in OP-IM shift instructions for encoding instructions, even in RV128.
(3) Potentially less operating system mode switching overhead in mixed 64/128 operation?  (64 bit code can simply call 128 bit code using a direct jump, no need to go into a syscall to change mode bits)
(4) Less pipeline complexity?, ie: don't need to handle flushing of invalid RV128 (but valid RV64) instructions from pipelines upon mode switch.  (Presumably RV32 would be more or less deprecated from systems capable of RV128 by such a future point in time).

cliffordwolf, 2019-08-14:
@xphung wrote:
> Below is the v0.90 BitManip proposed encoding of FSRI:
> `rs2 | 1 | imm6 | rs1 | func3 | rd | OP-IM`

@aswaterman wrote:
> I think @cliffordwolf will have to help with this discussion. I don't understand why rs2 isn't in the usual place, so it's hard to comment on the rest of question 1.

OP also opened [this issue](https://github.com/riscv/riscv-bitmanip/issues/42) in the riscv-bitmanip repository.

What they write about the rs2 position is a lie. I've clarified that before in the other issue but they still make this false claim. I have no idea what motivates them to make this stuff up. Here is a literal screenshot of the bitmanip spec document showing that the field on the left is `rs3`:

[image]

@xphung wrote:
> note there is a "cosmetic" renaming of rs2 into rs3 in his terminology

There is no "cosmetic renaming" going on here, whatever that's supposed to mean. The second operand of a funnel shift (`rs2`/`imm`) is the shift amount. It's in its usual place. The 1st and 3rd operand are the data bits. They come from `rs1` and `rs3`, both in their usual places.

I'm happy to discuss any actual design decisions in the bitmanip spec. But OPs interpretation is a strawman.

xphung, 2019-08-14:
> What they write about the rs2 position is a lie. I've clarified that before in the other issue but they still make this false claim. I have no idea what motivates them to make this stuff up. Here is [...]

The last line matches what I put at the start of this issue submission (aside from immaterial renaming of rs3 vs rs2), so I don't see how it's a "lie":

`
rs2 | 1 | imm6 | rs1 | func3 | rd | OP-IM
`

I highlight this issue because:
(1) @cliffordwolf directed me as follows "I'd direct you to the RV128 task group for that" when I raised the issue on GitHub's BitManip source page
(2) Creating a new R-type 2 source register + one destination register format is a major change for RISC V, and hard to justify for a single instruction (when the standard R-type format will do just fine).  It is very important for superscalar decoders to have register fields always in the same position within the instruction, and putting the second source register in a new position complicates instruction decode (just renaming it as rs3 instead of rs2 doesn't fix this)... this is very a basic RISC V principle, so I'm not really suggesting anything radical.

cliffordwolf, 2019-08-14:
> so I don't see how it's a "lie"

It's a falsehood because you claim its rs2 in an unusual location, when in fact it's rs3 in the usual location. And it's a lie because I've told you now a couple of times that this is incorrect, so you know by now. And knowingly telling a falsehood, that's lying.

> I highlight this issue because:
(1) @cliffordwolf directed me as follows "I'd direct you to the RV128 task group for that" when I raised the issue on GitHub's BitManip source page

Which you are now framing it as if the word "that" in my comment would have referred to the rs2/rs3 thing, which of course it didn't.

xphung, 2019-08-14:
Whether the register field is called rs2 or rs3, it doesn't change the fact that FSRI only has two register fields - it's immaterial what name the second register is given.

The existence of an immediate field as the 3rd input also doesn't really change much the decoding of the two source register fields.  (As an example, SD/SW/SH/SB instructions similarly have an immediate field as a 3rd input, but still keep the two source register operands in the "normal" positions).

The important thing is that superscalar instruction decoders (when decoding the two source register fields) can find the two source register fields in their expected location within the instruction word, so they can check for register out-of-order read/write conflicts as early as possible in the processing pipeline.  (This is very basic RISC microarchitecture knowledge, it's not anything I'm inventing myself - but it should have been a starting point principle when creating the BitManip instruction encoding - obviously it wasn't!  There are likewise other issues with the BitManip encodings, which I'll hold off bringing up for now, as it's important to settle this matter first).

xphung, 2019-08-14:
P.S. Please don't "shoot the messenger" (me) on me pointing out where I think designs can be improved.  I'm not trying to be unnecessarily difficult nor do I do this with any personal animosity, I just want to ensure RISC V's ISA design is the best possible.

I'd love to be more collaborative, but this got shut off early as a "time waster", without enough weight given to understanding the basic microarchitecture design issues, and I was redirected elsewhere, so that's what I did.

cliffordwolf, 2019-08-14:
I don't even know where to start. A pipeline will have existing logic for MUX'ing the imm field into the data path for rs2. rs3 is always in the same position. There is really no problem here. You are for some reason renaming rs3 into rs2 so that you can create the illusion of a problem. In fact, the encodings you proposed in the other issue would _create_ the need for all kinds of additional MUXes. I have no idea why you feel the need to waste everyones time like this. I'm now going to unsubscribe from this issue.

xphung, 2019-08-14:
Decoding a 6 bit immediate (even if in new position & needing MUX) is not the time critical path.
The time critical path is more likely to be decoding & scoreboarding/renaming the registers, so optimal positioning of register fields is more important than the immediate fields

xphung, 2019-08-17:
> In fact, the encodings you proposed in the other issue would _create_ the need for all kinds of additional MUXes.

The existing R4-type FMADD is instructive.

To avoid MUX'ing, a similar argument could have been made that the FMUL instruction should not have used the R-type instruction format - and instead use a format similar to your proposed FSRI (ie: not use an rs2 register placement, but instead place it's second source register in the rs3 position).

But RISC V designers never made that decision for FMUL, and potentially (depending upon exact implementation) instead went for a MUX up to 128 bits (for Quad FP).

For FSLI, the case to avoid MUX'ing is even weaker as we are talking about MUX'ing a maximum of 6 bits (for the shift amount) - a tiny fraction of what RISC V does for FMUL.

So bottom line is your excessive MUX argument was already rejected by the FMAD/FMUL precedent, and I don't see any reason to create a new precedent for FSRI.