riscv/riscv-isa-manual #438: How about adding push/pop instructions?

hyf6661669, 2019-08-21:
If we want to save the value of the general purpose registers that will be modified in a function, we have to execute instructions like:
```
li x2, STACK_POINTER
sw x1, 0(x2)
sw x3, 4(x2)
sw x4, 8(x2)
...
```
And when we are going to exit from this function, we have to execute instructions like:
```
li x2, STACK_POINTER
lw x1, 0(x2)
lw x3, -4(x2)
lw x4, -8(x2)
...
```

If we have push/pop instructions like what is described in ARM:
[image]

Then we only have to use 1 or 2 instructions to save/restore registers, which might be useful in  reducing the code size.

Since the push/pop instructions will make the hardware more complicated, we can add them as extended instructions.

hyf6661669, 2019-08-21:
Sorry, the push/pop instructions maybe make it more "CISC-like". Will the C compiler for ARM/X86 use push/pop instructions for the call/return of a function?

jim-wilson, 2019-08-21:
Depends on the core.  On some x86 cores, push/pop are slow, and they aren't used.  On some x86 cores, push/pop are fast, and they are used.  Likewise on arm, some cores prefer push/pop and some prefer ldrd/strd.  And note that while armv7 has push/pop armv8 does not.  If we have multiple ways to emit function prologues and epilogues, then there will be some additional complexity in the compiler to choose between which method is best.

Note that we have a gcc -msave-restore option for RISC-V which is supposed to serve the same effective purpose as push/pop instructions.  This calls functions for the register saves and restores, which reduces it to a call and maybe an auipc, plus the overhead of linking in the register save/restore library routines.  These functions can't be used in every function though.

hyf6661669, 2019-08-21:
@jim-wilson 
But many popular MCUs such as cortex M0, M3, M4, M7 support push/pop instructions. If we want to compete with ARM in IoT field, the larger code size could be a disadvtange.