riscv/riscv-isa-manual #442: Definition of `sstatus` and `vsstatus` unclear

Columbus240, 2019-08-22:
* Are changes to `mstatus` or `sstatus` always reflected in the other (if the change was to a field available in `sstatus`)? In other words: must an implementation behave as if `sstatus` were a restricted view of the fields in `mstatus`?
Sources: https://github.com/riscv/riscv-isa-manual/blob/5186fc7ab1dd113e49f86bd8342117263688bc52/src/machine.tex#L364-L365 https://github.com/riscv/riscv-isa-manual/blob/5186fc7ab1dd113e49f86bd8342117263688bc52/src/supervisor.tex#L173-L178

I’m not sure, because the second quote is only a commentary.

* What about `vsstatus`?
https://github.com/riscv/riscv-isa-manual/blob/5186fc7ab1dd113e49f86bd8342117263688bc52/src/hypervisor.tex#L601-L607
Must changes in `vsstatus` be reflected in `sstatus` or `mstatus`? It doesn’t look likely or necessary to me.

aswaterman, 2019-08-22:
I don't find it ambiguous, so maybe some other folks should weigh in.

Columbus240, 2019-08-22:
Do you mean, the commentary concerning `sstatus` should be interpreted as part of the specification and not as commentary?

aswaterman, 2019-08-22:
I meant that “a restricted view of mstatus appears in sstatus” seems to me to be sufficient to convey the idea, and that the commentary is merely clarification, and I was curious if other people disagreed with my interpretation.

commit 70b5dee5 (2020-03-04): Clarified description of sstatus.
Closes #442
closed 2020-03-04 by kasanovic