riscv/riscv-isa-manual #443: Figure A.15 outcome register value doesn't look correct

noahsherrill, 2019-08-29:
https://github.com/riscv/riscv-isa-manual/blob/66d31a74b98816dd756a0e00e3c208c51e18ddee/src/memory.tex#L759

The final value for a1 looks like it should be 1 rather than 0. The sw instruction at (f) immediately precedes the lw instruction at (g) in program order. (f) will always write 1 to the location at 0(s2), so I don't see any way that (g) could read anything else but 1 from the location at 0(s2). I don't see any other modifications to a1 in the listed instructions. Also, the diagram further appears to imply that (g) will read 1, as it indicates, "g: Rz = 1".

I apologize if this is simply a failure to understand on my part.

daniellustig, 2019-08-29:
Good catch @noahsherrill, thanks for pointing it out.  The typo is really that I got the register wrong, not the value...the fix should be to make it `{\tt a0=1}, {\tt a3=0}`.  As you say, `a1` is not really in question here.

aswaterman, 2019-08-29:
Fixed on master.  Thanks, guys.

commit d7d1c0c5 (2019-08-29): Fix outcome description for Figure A.15

h/t @noahsherrill with an assist from @daniellustig

Closes #443
closed 2019-08-29 by aswaterman