riscv/riscv-isa-manual #449: Behavior clarification is needed for pmpcfg.L==1 and pmpcfg.A==0

chuanhua, 2019-09-26:
In section 3.6.1 (PMP CSRs), Locking and Privilege Mode subsection, it is not clear what the behavior is when pmpcfg.L==1 and pmpcfg.A==0.

Could we add a sentence to clarify the behavior under this condition?

aswaterman, 2019-09-26:
What is unclear? The A and L fields are orthogonal. It’s just a PMP
register that is unmodifiable and matches no addresses.

chuanhua, 2019-09-27:
Why the A and L fields are orthogonal? When a PMP entry is off (A==0), the L field should have no effect, right? Otherwise, once a L bit is set, how can I turned on the entry to setup more protections?

chuanhua, 2019-09-27:
The reset values for L and A fields are not specified in the spec, so it is possible that after reset, L is set to 1 and A is set to 0.

aswaterman, 2019-09-27:
The Reset section specifies that A=0 and L=0 after reset: https://github.com/riscv/riscv-isa-manual/blob/efd8a03217843966d3e9a139c321b97d7725b1b6/src/machine.tex#L2462

chuanhua, 2019-10-01:
Thanks for pointing me to the reset section. I missed it. But, the important point is that at least it is not clear to me that the L and A field are orthogonal. Usually, from my experience, when the enable field of a table entry is disabled, the other fields in the entry will have no effect. So, could you clarify this point in the spec to reduce the possible misunderstanding about the relationship between the A and the L fields?

aswaterman, 2019-10-02:
I appreciate the feedback that this is unclear.  I will fix it.

commit 1d64f735 (2019-10-02): Clarify that pmpcfg.L takes effect even when pmpcfg.A=0

Resolves #449
closed 2019-10-02 by aswaterman