riscv/riscv-isa-manual #452: Clarify descriptions of LUI and AUIPC immediate values

robalni, 2019-10-11:
On pages 18-19 section 2.4 and 36-37 section 5.2 and 105 section 16.5, in the descriptions of LUI and AUIPC, I don't think it's clear what the "U-immediate value" and "nzimm" are.

    LUI (load upper immediate) is used to build 32-bit constants and uses the
    U-type format. LUI places the U-immediate value in the top 20 bits of the
    destination register rd, filling in the lowest 12 bits with zeros.

It places the "U-immediate value" in the top 20 bits. What is the "U-immediate value"? In the figure above the quoted text I can see:

```
 31                12 11   7 6        0
+--------------------+------+----------+
|    imm[31:12]      |  rd  |  opcode  |
+--------------------+------+----------+
          20             5        7
  U-immediate[31:12]   dest      LUI
  U-immediate[31:12]   dest     AUIPC
```

So it looks like the upper bits (31:12) of the "U-immediate value" are defined as the bits from the instruction encoding. The lower bits are still undefined.

Now it is not clear what the "U-immediate value" is.

* Is it a 32 bit value with the lower 12 bits undefined but bits 31:12 defined, as seen in the figure? That would mean that the value constructed in `rd` will be a 44 bit value (20 defined bits (imm[31:12]) followed by 12 undefined bits (imm[11:0]) followed by 12 zeros).
* Is it a 20 bit value that has no lower bits? But then how is it possible to have a value with no lower bits? That just doesn't make sense. The bit range 31:12 suggests that there should be bits 11:0 too.

---

I can see a few ways to make it clearer:

1. Always refer to the upper bits and never to the whole "U-immediate value". Example: change "LUI places the U-immediate value" to "LUI places the upper 20 bits of the U-immediate value" and "AUIPC forms a 32-bit offset from the 20-bit U-immediate" (on the next page) to "AUIPC forms a 32-bit offset from the 20 upper bits of the U-immediate". **This solution creates a problem mentioned at the bottom.**
2. Define the "U-immediate value" as a 32 bit value and define the lower bits of it as 0. Example: Change the whole sentence starting with "LUI places the U-immediate value" to "LUI places the U-immediate value, with the lower 12 bits set to 0, in the destination register rd", and a similar change for AUIPC.  **This solution creates a problem mentioned at the bottom.**
3. Define the "U-immediate value" as a 20 bit value (with bits 19:0) and change the figure to refer to "imm[19:0]" and "U-immediate[19:0]". That way it doesn't look like there are more than 20 bits in this value.
4. Define it as a 20 bit value but make it clearer somehow that 31:12 really refer to bits 19:0 of that value.

I think the intention is that it is a 20 bit value but the bits are numbered 31:12 instead of starting from 0, like all other numbers do, to show that this number will be used for upper bits. I think this just makes it more confusing though. This adds confusion when writing assembly code too because this bit numbering makes it look like you have to give really large values as arguments to lui and auipc (values that are >= 0x1000) to make them count as more than 0 since it looks like bits 11:0 will be ignored.

In chapter 25 when `auipc` is written in assembly code, it is not clear how the value given as argument to `auipc` (`delta[31:12]+delta[11]`) should be interpreted. It is given as a 20 bit number and that means that the "U-immediate" should probably not be interpreted as a 32-bit number because then these 20 bits would be bits 19:0 and there would be no way to reach all of the bits 31:12 of the "U-immediate".
 
Same in section 16.5 and the instruction `lui rd, nzimm[17:12]`.

aswaterman, 2019-10-11:
Figure 2.4 pictorially defines the U-immediate to be {inst[31:12], 12'b0} (using Verilog notation).

robalni, 2019-10-12:
Oh, now I think I understand how to interpret figure 2.4. I looked at that figure before posting this issue but it just looked like it was showing which bits in the instruction that were used and not how the immediate types are constructed.

So if I understand correctly, that means the U-immediate should be interpreted as a 32-bit number with the 12 lower bits = 0. This still has the problem I talked about at the bottom of my post. If the number is a 32-bit number and only the upper 20 bits are used (because the lower are 0) then assembly code like this:

```
lui rd, nzimm[17:12]
```

should only set the lower bits (which will not have any effect since they will be 0). That means that assembly code needs to give a very big number as argument to lui and auipc, something like this:

```
lui rd, nzimm[17:12]<<12
```

to reach bits 31:12, but that's not how I see most assembly code written.

Also, text like "LUI places the U-immediate value in the top 20 bits" makes it seem like the whole 32 bit value will be placed in the top 20 bits but that's obviously not possible.

aswaterman, 2019-10-18:
Sorry, somehow this slipped through the cracks.

The assembly syntax doesn't match the notation used in the ISA spec. This is unfortunate, but changing it would be too disruptive at this point.  Concretely, `lui rd, 0x1` writes the value `0x1000` to `rd`.

robalni, 2019-10-19:
First, what do you mean by "The assembly syntax doesn't match the notation used in the  ISA spec"? I mean that the assembly syntax used in the spec doesn't work well with the definition of the U-immediate because there is no explanation how small specified values will end up in bits U-immediate[31:12]. There is also the problem that some text suggests that a 32-bit value should be placed in the top 20 bits.

So I can see the following solutions to this:

- Change the definition of U-immediate to 20 bits instead of 32 and refer to U-immediate[19:0] instead of U-immediate[31:12]. 
- Change assembly syntax so you have to specify larger values. + Change text like "LUI places the U-immediate value in the top 20 bits". This should not be done because it will break existing code or make the assembly syntax incompatible with the one used by people.
- Add a description next to the definition of U-immediate that the specified value in assembly code is placed in the upper 20 bits of the U-immediate. + Change text like "LUI places the U-immediate value in the top 20 bits".
- Add a new section to the spec that explains the assembly syntax. + Change text like "LUI places the U-immediate value in the top 20 bits".

Do you think there is any good solution to this?

aswaterman, 2019-10-19:
Sorry for the confusion.  I was referring to syntax used by the GNU/LLVM assemblers, which the spec should also strive to match.

I will make a PR that fixes this in an expedient way.

closed 2020-03-04 by kasanovic

Chathura-De-Silva, 2023-08-27:
> 

I got in to the same confusion and finally found this.

hackGeek-04, 2024-06-01:
Thank you for the clarification

commit 0ac0b9a8 (2019-10-19): Consistently claim that U-immediate is 32 bits, not 20 bits

The text vacillates between describing the U-immediate as a 20-bit
quantity, and as a 32-bit quantity whose lower 12 bits are zero.
Standardize on the latter.  Note the discrepancy in the ASM syntax.

Resolves #452
PR #453 (2020-03-04): Consistently claim that U-immediate is 32 bits, not 20 bits
The text vacillates between describing the U-immediate as a 20-bit
quantity, and as a 32-bit quantity whose lower 12 bits are zero.
Standardize on the latter.  Note the discrepancy in the ASM syntax.

Resolves #452