riscv/riscv-isa-manual #456: Why is the value of htimedelta added to time in VS mode or VU mode?

tfcata, 2019-10-30:
`htimedelta` seems to be a unsigned integer, adding it to `time` will only produce a time value in the future. But usually time offsets are used to adjust time view in the VM so that the time a VM gets is always in the past. (AFAIK, both ARM and x86 substract a positive value from the real time value.) Why is the time offset mechanism defined differently in here?

aswaterman, 2019-10-30:
It’s modular addition, so it’s equivalent to subtraction. If you want to subtract N, just write -N to the delta register.

tfcata, 2019-10-30:
I find writing -N rather unintuitive. Effectively it causes an overflow in ˋtimeˋ. And using an overflow for normal operation like adjusting time for a VM is rather 'hack-ish', so to speak. This would be marked as an exceptional case for any code scanner that scans for possible overflow bugs, for example.

Any specific consideration behind this decision?

gfavor, 2019-10-30:
Maybe a better way to view this issue is the following:  The spec
should explicitly say that htimedelta is a signed value.  Then there is no
"overflow" possibility for a code scanner to point the finger at.  And I
suspect the answer to why htimedelta was defined as being _added_ to time
is because this is the simplest straightforward definition.  A desired
negative time delta would naturally be expressed as a negative number.

tfcata, 2019-11-12:
While it is certainly helpful for the spec to clearly say that `htimedelta` is a signed value, constructing a negative value from unsigned values read from the clock sources such as cycle counters still require extra steps in software. I have the impression that most of the time the value read by a VM is in the past compared to real time, so by specifying 'a signed value (most of the time negative) to be added to `time`', the 'normal' path would involve more operations then the 'abnormal' path. 

An alternative is to move the negating logic into hardware so that the complexity above is reversed, on the 'normal' path an unsigned value is written to htimedelta, meanwhile on the 'abnormal' path more operations are needed.

aswaterman, 2019-11-12:
Can someone research how Intel64, AMD64, ARMv8, POWER, etc. represent this?  If they all agree on a certain polarity, it makes sense to follow suit to avoid software bugs.  If some go one way and some go the other, we can just stick with the status quo.

gfavor, 2019-11-12:
ARMv8 says "The virtual counter contains the value of the physical counter
minus a 64-bit virtual offset.".

Intel64 says "The value of the TSC offset is added to the value of the
time-stamp counter.".  AMD64 not surprisingly indicates the same thing.

Someone else will need to provide an answer for POWER and other
architectures.

Greg

aswaterman, 2019-11-12:
Thank you, @gfavor.  To me, that is sufficient evidence that the choice is fairly arbitrary, and conveniently our current choice matches the dominant architecture in this space.

tfcata, 2019-11-12:
OK. In this case please still state that `htimedelta` is signed.  That avoids some confusion.

aswaterman, 2019-11-12:
OK, thanks for raising the issue, @tfcata; I have added a note in b25d934b89f0b14a35f80c83469af24b8326ad43 (with @jhauser-us to review).

closed 2019-11-12 by aswaterman