riscv/riscv-isa-manual #458: M-mode interrupt clarification

eak, 2019-11-14:
Some of the engineers I’m working with have found the phrase “M-mode interrupts” to be confusing. It only occurs in one place, in the sentence below:

By default, M-mode interrupts are globally enabled if the hart’s current privilege mode is less than M, or if the current privilege mode is M and the MIE bit in the mstatus register is set.

Might I suggest the following rewrite of the paragraph to define things, presuming that my rewrite is correct:

Each interrupt may be delegated to a less privileged mode. If an interrupt is not delegated in mideleg, it is an M-mode interrupt. If it is delegated in mideleg and not delegated in sideleg it is an S-mode interrupt. If it is delegated in both mideleg and sideleg then it is an U-mode interrupt. An interrupt i will be taken if bit i is set in both mip and mie, and if interrupts are globally enabled for x-mode, the delegated mode of interrupt i. Interrupts are globally enabled if the hart’s current privilege mode is less than x-mode, or if the current privilege mode is equal to x-mode and the xIE bit in the mstatus register is set.

There was also some confusion about SSIP, which seemed to be named assuming it would be a S-mode interrupt, but which is actually an M-mode interrupt unless so delegated with mideleg. Is this worth a note?

aswaterman, 2019-11-14:
Understood.  Rather than adding text to define the term "M-mode interrupt", can we just avoid using that term, perhaps by instead writing "interrupts destined for M-mode", which conveys we're talking about interrupts that have not been delegated?

So, we could change the sentence to say, "Interrupts destined for M-mode are globally enabled if the hart’s current privilege mode is less than M, or if the current privilege mode is M and the MIE bit in the mstatus register is set."

eak, 2019-11-14:
Avoiding definition of a new term is good, but I think the paragraph is still a bit hard to interpret because of the "by default" and "however". While the following is wordier, I think it is clearer. What do you think?

An interrupt i will be taken if bit i is set in both mip and mie, and it is globally enabled for its target mode. Each interrupt may be delegated to a less privileged mode. If interrupt i is not delegated in mideleg, then it is globally enabled if the hart’s current privilege mode is less than M, or if the current privilege mode is M and the MIE bit in the mstatus register is set. If it is delegated in mideleg and not delegated in sideleg then it is globally enabled if the hart’s current privilege mode is less than S, or if the current privilege mode is S and the SIE bit in the mstatus register is set. If it is delegated in both mideleg and sideleg then it is globally enabled if the UIE bit in the mstatus register is set.

Rolfy360, 2019-12-09:
BTW, in Section "Supervisor Interrupt Registers (sip and sie)", user can not see any controllable bits unless they are delegated from M-mode. May we can have them all explained in MIP/MIE section and make reference in SIP/SIE section.

peiyuanix, 2023-07-05:
Don't understand why the `mstatus` includes `SIE`, `SPIE` and `SPP`?

kersten1, 2024-03-28:
After much search around, I think this one has been completed. I'm closing for now. Please reopen if you disagree.

closed 2024-03-28 by kersten1