riscv/riscv-isa-manual #461: in User-Level ISA Specification.pdf

lizhirui, 2019-12-04:
in User-Level ISA Specification.pdf
page 127,Table 16.7(Instruction listing for RVC,Quadrant 2)
near "C.EBREAK","inst[11:7] = 0 but inst[6:2] != 0" reserved?  that wasn't written here,undefined

scottj97, 2020-01-07:
I'm not sure I understand, but I think you're saying there should be another row in the table below C.EBREAK with columns:

100 | 1 | 0 | - | 10 | Reserved

But this case is covered two rows later under C.ADD where it says "(HINT, rd=0)". I would say it's a bit confusing that the table row for C.ADD says "rs1/rd != 0" even though the case of rd=0 is in fact covered by that row. Same goes for several other similar rows in this table.

I confirmed that Spike does not take any illegal instruction exception for those opcodes, instead decoding them as C.ADD to x0. Sail is taking traps but there's a [PR to change that](https://github.com/rems-project/sail-riscv/pull/31/files).

kasanovic, 2020-03-04:
The case is covered as described above.
But it is noted that the C instruction listing could be expanded to be clearer.

closed 2020-03-04 by kasanovic