riscv/riscv-isa-manual #462: Some FCVT instructions missing frm as source register in dep-table.tex

acw1251, 2019-12-05:
It looks like the FCVT.L[U].S and FCVT.S.L[U] instructions are missing frm as a source register (with the note "*if rm = 111"). Here are the lines in question: https://github.com/riscv/riscv-isa-manual/blob/master/src/dep-table.tex#L368-L374

acw1251, 2019-12-05:
And also, does FCVT.D.W[U] raise an exception if there is an illegal rounding mode? If so, should it also have a dependency on frm, even though it produces exact results?

aswaterman, 2019-12-05:
For simpler HW, `FCVT.D.W[U]` still checks the validity of the rounding mode.  I don't think this implies a dependence on `frm` in the PPO sense, though, because it only affects the exception behavior of the instruction, not its result; and the possibility of taking an exception doesn't create a PPO dependence from later instructions.

`FCVT.L[U].S` and `FCVT.S.L[U]` are bugs in the table, since they definitely round.  @daniellustig I'm just going to update the table, since this is presumably just an uncontroversial bugfix.

aswaterman, 2019-12-05:
Thanks for the attention to detail, @acw1251.

aswaterman, 2019-12-05:
Closed by 015acc5602587d9662c6b2db4685ae8b06d577d4

closed 2019-12-05 by aswaterman