riscv/riscv-isa-manual #466: About LR/SC sequences

abdullahyildiz, 2019-12-13:
According to section 8.2 of _The RISC-V Instruction Set Manual Volume I: Unprivileged ISA Document (Version 20190608-Base-Ratified)_, it says that "SC.W writes zero to rd on success or a nonzero code on failure."

Should the following instruction cause an exception according to the related spec since rd is the zero register:

`sc.w zero, ra, (s11)`

kasanovic, 2019-12-13:
No.  This form is useful for clearing any existing reservations without destroying a register.

closed 2019-12-13 by kasanovic

abdullahyildiz, 2020-08-11:
@kasanovic,

What is the correct behavior when the reservation is made for single word by LR but SC writes doubleword to the address in rs1 (or vice versa)? Should SC succeed or not?

sorear, 2020-08-14:
@abdullahyildiz From §8.3 Eventual Success of Store-Conditional Instructions:

> The SC must be to the same effective address and of the same data size as the latest LR executed by the same hart.

So if the size changes, the SC is not required to succeed, although it may if the LR creates a reservation set large enough.

abdullahyildiz, 2020-08-14:
@sorear,

Thanks for the explanation.