riscv/riscv-isa-manual #469: Possible ambiguity regarding AMO exceptions

jrtc27, 2019-12-23:
The spec defines two different causes: `Load access fault` (5) and `Store/AMO access fault` (7). AMOs are both loads and stores. I assume the intent that AMOs always generate `AMO access fault` regardless of the fact that the load is (always?) the part that traps? Nowhere in the unprivileged and privileged specs can I find something explicitly stating this. Spike generates an `AMO access fault`, but currently the Sail model generates a `Load access fault` and I would like to ensure that this behaviour is not intended to be permitted before correcting it.

aswaterman, 2019-12-23:
As the name implies, AMOs generate AMO access faults. The SAIL model should
be changed.

FWIW, this is potentially better for performance, as it gives the OS more
accurate info about why the fault was caused. In some cases, a load fault
might bring the page in read-only, resulting in two faults.

aswaterman, 2020-03-02:
Closed via d6f078bfd0dd962d371c59960ca8b1dcaaf050a8

closed 2020-03-02 by aswaterman