riscv/riscv-isa-manual #479: Some abbreviations are not explained in priv spec

stefanct, 2020-01-27:
I noticed that some of the abbreviations used in the privileged specification are not written out fully. They are, however, defined and written out in the unprivileged/main spec.
A (very likely) incomplete listing of them:

- load-reserved (LR)
- store-conditional (SC)
- (the combination of the above, i.e. Load-Reserved/Store-Conditional (LR/SC)
- atomic memory operation (AMO)

I am not sure if it's on purpose to not repeat the full writing of abbreviations defined by the unprivileged spec in the privileged spec again but it would make the documents more accessible if they were and it's cheap to do.

aswaterman, 2020-03-04:
I scanned the A-extension chapter, and it looks like these acronyms are implicitly defined the first time they are used, e.g., `load-reserved (LR)`.

closed 2020-03-04 by aswaterman

stefanct, 2020-03-04:
Yes, they are - as I have explained above - written out in the main spec.
However, they are never written out in the privileged spec, which is a nuisance since it is a separate document.