riscv/riscv-isa-manual #480: definition of machine time compare register event

konrad-schwarz, 2020-01-29:
In V1.12-draft, the logic for the timer event is "mtime < mtimecmp" using an unsigned comparison.  The specification does not say what happens when mtime overflows.

The alternative definition "0 <= (mtime - mtimecmp)" using signed comparison would allow for the timer to overflow, thus allowing the system to run indefinitely.  The cost is that the maximum timeout period can only be half of the precision of the timer register (so 2^63 ticks) and that a true subtraction circuit rather than digital comparator are needed.

closed 2020-03-04 by kasanovic

konrad-schwarz, 2020-03-05:
Change [ad315fe](https://github.com/riscv/riscv-isa-manual/commit/ad315fe5812046abbf77f34580b5351bed54a3a4) raises the question what the mtimecmp register should do when mtime overflows.  Should it reset or stay set?

The advantage of the proposed definition for the timer event in #480 above is that it continues to work in the face of mtime overflows.

If you don't want to use the definition given in #480, I think it would be better to note that with a tick period of 1 ns, the 64-bit mtime will overflow after 584 years and hence the standard does not mandate a behavior.

aswaterman, 2020-03-05:
It’s a level, not an edge—the interrupt is pending whenever mtime >=
mtimecmp, so in most overflow scenarios, the interrupt goes low on
overflow, unless mtimecmp=0.

konrad-schwarz, 2020-03-05:
So on overflow, the interrupt is (silently) retracted -- obviously an undesirable outcome.

Note that the unprivileged spec says, in 10.1:
> The underlying 64-bit counter should
> never overflow in practice.
Why not have the privileged spec use the same approach?

zetalog, 2023-05-31:
IMO, 

The spec just says:
(unsigned)N > (unsigned)O means N is a newer timestamp, and O is an elder timestamp.
N/O is unsigned value, > is (greater than).
It says nothing about delta.
It depends on the implementation to comprehend.

Say:
(unsigned)N > (unsigned)O
can be converted into:
(unsigned)N - (unsigned)O > 0

The spec doesn't say something about delta value:
(unsigned)((unsigned)N - (unsigned)O) > 0
Have we seen a definition forcing that the delta must be unsigned? 

So why bother?
The implementation apparently can consider a wrap-over safe implementation w/o violating the spec.

commit ad315fe5 (2020-03-04): Clarified overflow behavior of mtime register.
Closes #480.