riscv/riscv-isa-manual #481: Is `htimedelta` mandatory?

palmer-dabbelt, 2020-01-30:
From reading the draft H extension it sounds to me like all 64 bits of `htimedelta` must be implemented if the H extension is implemented.  This seems like unnecessary CSR state: on systems that are going to trap/emulate timer accesses anyway (via the counter disable bits) the guest's time offset can be stored in memory instead.

Making it mandatory makes my life easier, so I don't really care that much either way.  I just wanted to make sure it's actually that way.

avpatel, 2020-01-30:
The HTIMEDELTA CSR is tide with TIME CSR because it controls
TIME CSR value read from VS-mode and VU-mode.

TIME (VS-mode or VU-mode) = TIME (HS-mode or M-mode) + HTIMEDELTA

If HW does not implement TIME CSR then HW should not implement
HTIMEDELTA as well. In this case, OpenSBI will trap-n-emulate both
TIME CSR and HTIMEDELTA CSR. If HW implements TIME CSR then HW
has to implement HTIMEDELTA as well.

To summarize, TIME CSR is optional for HW hence HTIMEDELTA is
optional for HW as well. Also, HW designers should either implement
both CSRs or skip both.

aswaterman, 2020-01-30:
Implementations that trap into M-mode for access to the time CSR would
presumably trap accesses to htimedelta in the same manner, so there should
be no hardware cost for such designs.

For implementations that don’t trap accesses to the time CSR, this is an
important feature, and the cost seems justified.

Is there another class of implementation you’re thinking of that doesn’t
fit into one of these buckets?

aswaterman, 2020-03-04:
Consensus via silence--closing :)

closed 2020-03-04 by aswaterman