riscv/riscv-isa-manual #485: Is address translation/paging in effective at machine mode?

chillancezen, 2020-02-14:
hi experts, 

as far as I know, the `satp`  register is controlling mmu address translation behavior. 
specifically in supervisor mode: if we set the 31th bit to 1, the paging is enabled.

what if an interrupt/exception  is taken back in m-mode later? does m-mode still use address translation to address memory, since `satp` is not touched?

thanks,
Link

chillancezen, 2020-02-14:
if my understanding is right, in m-mode, addressing is always flat, no address translation is performed.

in s-mode/u-mode, if paging is enabled in csr `satp`, address translation is always performed.

could someone help correct me please?  I find no answer for this from riscv-spec 

thanks,

kasanovic, 2020-02-15:
M-mode does not use address translation by default, but the memory privilege bits in the mstatus register can change the effective translation as described in that section of the manual.

The manual text could be improved to make this clearer, which is why I'm leaving this issue open.

allenjbaum, 2020-04-15:
To be more clear, that would be mstatus.MPRV together with .SUM, .SBE, MXR (if MPP==Smode and paging is enabled). Normally, M-code isn't expected to have exceptions, but if MPRV=1, then it could (if that access would have had an exception had the mode been MPP).
Mcode must make provisions for that by saving trap state before executing code with MPRV=1, and restoring it before returning. If I'm correct, that's another thing you might want to make clear.

closed 2020-05-18 by chillancezen

cheofusi, 2023-12-06:
Hi everyone, 

Consider the case where in M-mode, MPRV=1, and MPP=11. Should address translation be the M-mode default i.e no translation, or the effective translation, implying a recursive definition?

An example would be running in S-mode, trapping into M-mode, and setting MPRV=1. Now address translation is the way it was before the trap. Suppose we now set MPP=11. Do we continue using the page tables?