riscv/riscv-isa-manual #486: how does satp affects program counter?

chillancezen, 2020-02-17:
hi experts,

when I enabled paging in S-mode by writing `satp`, how the address of the next instruction is going to be impacted?

e.g. we have the instruction sequence below before paging and we have PC-relative addressing:
```
80000134:   18061073            csrw    satp,a2
80000138:   12000073            sfence.vma
8000013c:   00008067            ret
```

paging linearly maps `0x80000000~` to  `0xc0000000`, thus we have counterpart addresses:
```
c0000134:   18061073            csrw    satp,a2
c0000138:   12000073            sfence.vma
c000013c:   00008067            ret
```

the expected behavior for PC is jumping from `0x80000134` to `0xc0000138`, but how could that happen? (`we can not walk through the whole page table to find the virtual address of pa:0x80000138` )

thanks,

chillancezen, 2020-02-17:
this is so crucial  because I find right after `csrw satp`, there are PC-dependent instructions `auipc/lui` following.  wrong PC results in wrong behavior eventually. 
```
c000011c:   18051073            csrw    satp,a0
c0000120:   00000517            auipc   a0,0x0
c0000124:   06c50513            addi    a0,a0,108 # c000018c <relocate+0xb4>
c0000128:   10551073            csrw    stvec,a0
c000012c:   005ea197            auipc   gp,0x5ea
```

the `gp` is completely wrong if PC is not updated accordingly after  instruction `csrw    satp`.

aswaterman, 2020-02-17:
Writing satp does not affect the PC, aside from the usual += 4.

So when you enable virtual memory, the sequential instruction address must
be mapped in the page table you loaded, or you’ll take a page fault at that
point.

chillancezen, 2020-02-18:
thanks @aswaterman, 

```
when you enable virtual memory, the sequential instruction address must
be mapped in the page table you loaded
```
do you mean VA==PA after paging is enabled? 

thanks.

aswaterman, 2020-02-18:
Yeah, creating an identity mapping so that VA==PA is the most straightforward way to accomplish that.

chillancezen, 2020-02-18:
oh, thanks Andrew, my issue in my hypervisor is perfectly resolved by raising an `instruction paging fault`.

here is how Linux relocates itself from 0x80040000 to c0000000.

```
c00000d8 <relocate>:
c00000d8:   c00005b7            lui a1,0xc0000
c00000dc:   00000617            auipc   a2,0x0
c00000e0:   f2460613            addi    a2,a2,-220 # c0000000 <_start>
c00000e4:   40c585b3            sub a1,a1,a2
c00000e8:   00b080b3            add ra,ra,a1
c00000ec:   00000617            auipc   a2,0x0
c00000f0:   03460613            addi    a2,a2,52 # c0000120 <relocate+0x48>
c00000f4:   00b60633            add a2,a2,a1
c00000f8:   10561073            csrw    stvec,a2
c00000fc:   00c55613            srli    a2,a0,0xc
c0000100:   800005b7            lui a1,0x80000
c0000104:   00b66633            or  a2,a2,a1
c0000108:   005ed517            auipc   a0,0x5ed
c000010c:   ef850513            addi    a0,a0,-264 # c05ed000 <trampoline_pg_dir>
c0000110:   00c55513            srli    a0,a0,0xc
c0000114:   00b56533            or  a0,a0,a1
c0000118:   12000073            sfence.vma
c000011c:   18051073            csrw    satp,a0
c0000120:   00000517            auipc   a0,0x0
c0000124:   06c50513            addi    a0,a0,108 # c000018c <relocate+0xb4>
c0000128:   10551073            csrw    stvec,a0
c000012c:   005ea197            auipc   gp,0x5ea
c0000130:   68c18193            addi    gp,gp,1676 # c05ea7b8 <netdev_budget_usecs>
c0000134:   18061073            csrw    satp,a2
c0000138:   12000073            sfence.vma
c000013c:   00008067            ret
```

any exceptions will be trapped to 0xc0000120(which is correctly set in `mtvec` csr.) what's is interesting is at the end of the code snippet, there is an instruction `ret`, not `sret`. 
 
it's great to know how this works. 

thanks again.

aswaterman, 2020-02-18:
Ah yes, I wrote that code a long time ago.  Glad it's working for you, too.

closed 2020-02-18 by aswaterman