riscv/riscv-isa-manual #487: Enabling N-extension registers.

michael-roe, 2020-02-24:
Section 6;3 says:

"An NS field can be added to mstatus and sstatus following the format of the FS and XS fields
to reduce context-switch overhead when the values are not live."

a) This doesn't say which bits in mstatus contain the NS field.

b) If I understand it correctly, NS disables access to uepc, ucause and utval but does not affect uscratch or utvec. It might be better if NS  enabled/disabled all of them, so that operating system context switch does not need to save/restore any of these registers for processes that aren't using user-mode exceptions.

P.S. I undestand that you'te thinking of deprecating all the N-extension stuff.

aswaterman, 2020-02-24:
Hello again.  If we were moving forward with the N extension at this time, we'd probably put the NS field in bits 24:23.  For your purposes, the N extension is effectively a nonstandard extension anyway, so you could either allocate bits 24:23 as a nonstandard extension, or you could skip the NS field and leave the N extension unconditionally enabled.

aswaterman, 2020-03-04:
Closing, as N extension is deliberately not completely specified yet.

closed 2020-03-04 by aswaterman