riscv/riscv-isa-manual #488: Does CVT.W.S use convertToInteger or convertToIntegerExact?

pdonahue-ventana, 2020-02-29:
IEEE 754-2008 says:
> When the value of the conversion operation’s result differs from its operand value, yet is representable in the destination format, some conversion operations are specified below to signal the inexact exception and others to not signal the inexact exception.

It then defines several `convertToInteger` functions and similar `convertToIntegerExact` functions which differ only in their behavior with respect to the inexact exception.

Do RISC-V fp-to-int conversions such as CVT.W.S use the `convertToInteger` or `convertToIntegerExact` functions?  I think it's the latter but the spec doesn't say.

aswaterman, 2020-02-29:
These instructions raise the inexact exception.  I've added statement to that effect: 27b40fbc798357fcb4b1deaba4553646fe677576

pdonahue-ventana, 2020-02-29:
Looks good.  Thanks.

closed 2020-02-29 by pdonahue-ventana