riscv/riscv-isa-manual #489: GAS instruction syntax is missing for many instructions

konrad-schwarz, 2020-03-04:
Although an ISA specification, the RISC-V ISA Manual also uses a concrete assembler syntax, namely that of the GNU Assembler GAS.

However, the document is incomplete in this respect, e.g., how are the "aq" and "rl" bits in the LR/SC instructions to be coded?  Experimentation shows that GAS uses lr.d.aqrl, but it is unfortunate that such experimentation is required.  The GAS manual and the "RISC-V Asssembly Handbook" do not  provide any insight here.

The GAS manual is also missing the specification of the line comment char (`#'), whether a line separator character exists, how integer literals are notated (without any prefix or suffix character) and the byte lengths of the .int, .short, .long, and .word directives.  See the "XXX Syntax" chapters of GAS's other supported architectures for a template. 

When writing code designed to be compatible with 32 and 64-bit ISAs, it would be useful if there were a standard pseudo-op for loading and storing an xlen wide register and for loading an unsigned 32-bit value.

aswaterman, 2020-03-04:
We have decided the assembler manual should be a separate spec. It is a work-in-progress here: https://github.com/riscv/riscv-asm-manual

closed 2020-03-04 by aswaterman

konrad-schwarz, 2020-03-05:
Should I resubmit the issue there?

That manual does not include the information either.  I see that it is in an early state, but perhaps it would useful as a reminder.

aswaterman, 2020-03-05:
Yes, or make a PR that fixes the issue. (This is of course well defined already within GAS, just not written down in the spec.)