riscv/riscv-isa-manual #491: Add new section in preface for 20191213-Base-Ratified

kito-cheng, 2020-03-17:
There is latest ratified version for 20191213 https://github.com/riscv/riscv-isa-manual/releases/tag/Ratified-IMAFDQC

but there is no corresponding preface section like `Preface to Document Version 20190608-Base-Ratified` for 20191213.

I think we should add that to easier track ratified version?

aswaterman, 2020-03-17:
What we're doing is our standard pattern.  The released version has the main "Preface", and the older versions have "Preface to Document Version XXX".  When we release the next version, we will create a "Preface to Document Version 20191213-Base-Ratified", which contains the changes between 20190608 and 20191213.

kito-cheng, 2020-03-17:
Hmm, I saw last time for 20190608 we've made a preface page when bumping version in cover, the problem to me is I can't found 20191213 has a release in latest draft pdf.
https://github.com/riscv/riscv-isa-manual/commit/1e8c2e1fc2c1fd4a5b9177869aee8ace4b5879bb

aswaterman, 2020-03-17:
I see what you mean.  We should have created a similar commit after releasing 20191213.  I'll do that now.

commit 274893e2 (2020-03-17): Add preface section for older version 20191213

Closes #491
closed 2020-03-17 by aswaterman