riscv/riscv-isa-manual #492: Bitfield in expended instruction in C extension chapter is a bit misleading

takahirox, 2020-03-23:
Hi all. I made [RISC-V software emulator in Rust](https://github.com/takahirox/riscv-rust) and also compile it to WebAssembly so that you can also run [the RISC-V port xv6 (UNIX V6 rewritten by MIT for x86 in ANSI C) on your browser](https://takahirox.github.io/riscv-rust/index.html).

While I was implementing C extension in the emulator, I thought that bitfield in the C extension chapter in the manual may be a bit misleading.

For example, in the explanation of C.ADDI4SPN

>nzuimm[5:4|9:6|2|3] <= C.ADDI4SPN instruction [12:5]
>
>C.ADDI4SPN is a CIW-format instruction that adds a zero-extended non-zero immediate, scaled
by 4, to the stack pointer, x2, and writes the result to rd′. This instruction is used to generate
pointers to stack-allocated variables, and expands to addi rd′, x2, nzuimm[9:2].

I think `addi rd′, x2, nzuimm[9:2]` in this section may be a bit misleading. I don't think `nzuimm[9:2]` represents `scaled by 4` well because as of me `nzuimm[9:2]` is extraction of `nzuimm[9:0]`. For example, if `nzuimm[9:0]` is `0b11_1111_1100 (=x3fe)` then `nzuimm[9:2]` will be `0b1111_1111 (= 0xff = 0x3fe >> 2)`. 

Another example. If `nzuimm[9:2]` in C.ADDI4SPN instruction is `0b1111_1111` then the instruction should be expanded to `addi rd′, x2, 0xb11_1111_1100`. But from the expanded instruction representation `addi rd′, x2, nzuimm[9:2]` in the current C extension chapter manual, readers may misread that it would be expanded to `addi rd′, x2, 0xb1111_1111`.

So I'd like to suggest to rewrite it to either one to clarify how the instruction will be expanded

- `addi rd′, x2, nzuimm[9:0]` (implying nzuimm[1:0] is zero)
- `addi rd′, x2, nzuimm[9:2] * 4`
- `addi rd′, x2, nzuimm[9:2] << 2`

In the C extension chapter, other instructions explanation have the same problem. If you agree with my opinion I'd like to make PR.

aswaterman, 2020-03-23:
Or maybe the simpler option:

addi rd′, x2, nzuimm

takahirox, 2020-03-23:
I like that. I will make PR soon.

aswaterman, 2020-04-10:
Resolved by #506

closed 2020-04-10 by aswaterman