riscv/riscv-isa-manual #494: Considerations for an RV32EG

jrmoserbaltimore, 2020-03-25:
The RV32/64G extensions define a standard baseline, that being RV32I or RV64I plus MAFD, Zicsr, and Zifencei.

Would it make sense to add an RV32EG replacing FD with Zfinx when drafted?

I'm particularly considering FPGA implementations, from small (2,000LE) to larger (80,000LE) FPGAs, both for embedded microcontrollers and more general-purpose RISC-V Soft CPUs.  I selected the above extensions because:

* Modern FPGAs provide hardware multipliers, so the M extension has a small footprint
* Single-hart implementations can implement atomic instructions by decoding to equivalent non-atomic instructions (I think?)
* Zfinx will (assumedly) avoid a bunch of floating point instructions and registers, saving a great deal of space while still providing floating-point operations

I imagine these will provide a significantly-smaller implementation for FPGA by eliminating F and D instructions in favor of Zfinx; while for ASIC, the RV32E implementation saves a ton of space, and Zfinx extension provides valuable floating-point without adding a heavy register file.  It remains to be seen if the Zfinx instruction implementation will be substantially more space-efficient than F.

I imagine Linux and Minix could be induced to run on these with minimal effort.

One consideration:  if using Zfinx, then the code is not compatible with RV32G; whereas RV32E-MAFD-Zicrs-Zifencei would run on RV32G.  Should future revisions of RV32G specify Zfinx?  Will Zfinx provide some performance advantages?  (For that matter, the concept of Zfinx—reusing the integer registers as floating-point registers—seems to reduce the instruction count and processor complexity in general.  I might have preferred it to FD, had Zfinx and Zdinx been drafted before RV32/64G; but what do I know anyway?)

aswaterman, 2020-03-25:
To avoid confusion, we should avoid reusing the G moniker.  In the long run, ISA names are going to get long.  Perhaps compressing IMAFD into G was a false economy to begin with.

RV32EMAFCZfinx is probably going to be a popular design point.  Adding D implies adding the concept of values spanning two x-registers--not inherently problematic, but certainly a new thing to deal with.

I'm a little surprised this is coming up in the context of a soft core.  The main cost of I vs. E, or not-Zfinx vs. Zfinx, is regfile capacity, which should not command a premium cost in FPGAs.  If you're building deeply threaded cores, I can understand the desire to reduce regfile capacity, though.

jim-wilson, 2020-03-25:
There is no toolchain support yet for rv32ef or rv32efd.  There is also a problem that the current rv32e abi can't support rv32efd correctly because d requires 64-bit alignment and the rv32e abi requires 32-bit stack alignment.  This is one of the reasons why the toolchain support hasn't been implemented yet.  To fix this, we need a new rv32e abi.  There are plans for a new embedded abi that would fix the problem, but those plans have been stalled for about 2 years now and not clear when the work will continue.

There is also no toolchain support for Xfinx yet.  This also requires ABI changes, but I think they are simpler than the new embedded abi.  There was some discussion about defining a Zfinx ABI recently, so this may happen before the new embedded abi.  But this doesn't solve the alignment issue for doubles, so there may still be a problem, depending on whether there is still a load double instruction and whether it still requires 64-bit alignment.

if your timelines are short, you might want to pick an architecture combination that already has toolchain support.  if you timelines are long, there is a chance the toolchain support might be written before you need it.

jrmoserbaltimore, 2020-03-28:
> To avoid confusion, we should avoid reusing the G moniker.

Fair enough.  I guess the embedded implementation isn't a common-target thing anyway.  My thinking was mostly to supply the same facilities, although Zfinx isn't in IG and so RV32EG as described wouldn't run on RV32IG (which might not be useful anyway).
 
> RV32EMAFCZfinx is probably going to be a popular design point.
> 
> I'm a little surprised this is coming up in the context of a soft core. The main cost of I vs. E, or not-Zfinx vs. Zfinx, is regfile capacity, which should not command a premium cost in FPGAs. If you're building deeply threaded cores, I can understand the desire to reduce regfile capacity, though.

I'm thinking in terms of both soft and hard cores, although floating-point is huge—a small RV32I costs 700 LUTs on a Xilinx FPGA, while I've seen low-area single-precision floating-point multiply-and-accumulate boast such small areas as 250-275 4-input LUTs.  The floating-point logic would be
larger than the CPU itself.  To be fair, they implemented an adder and a multiplier; and the multiplier they used (Nikhilam) may or may not be the most efficient (I've seen a Paravartya 8-bit integer divider in 72 LUTs and it was twice as fast as a 200-LUT Nikhilam divider and had half the delay—and apparently 80% of the delay of a typical reference divider).

RV32EMAZfinx might be sensible if float is needed in ASIC because it cuts back on all that register space, which you're going to need if you implement floating-point operations.  Not sure F would be interesting in that context, nor C (ROM space is cheaper than ASIC space, right?). Integer multipliers and dividers are cheap.

Anyway all this was probably pointless.

jrmoserbaltimore, 2020-04-17:
Thinking back, this discussion was probably ill-considered and can be closed.

closed 2020-04-17 by aswaterman