riscv/riscv-isa-manual #495: RVWMO, single-copy atomic, unaligned access emulatoin

Dolu1990, 2020-03-26:
Hi,

From the spec : 

> Memory-accessing instructions give rise to memory operations. A memory operation can be either
a load operation, a store operation, or both simultaneously. All memory operations are single-copy
atomic: they can never be observed in a partially-complete state.

To my understanding it seem incompatible with hardware which do not support unaligned access and emulate them in let's say machine mode via multiple bytes access.

If my understanding is right, maybe that rule should be weakened by only applying it to alligned access ?

aswaterman, 2020-03-26:
Sec. 6.1 says, "A misaligned load or store instruction may be decomposed into a set of component memory operations of any granularity"

So, the spec does say what you want it to say.

Dolu1990, 2020-03-26:
@aswaterman Ahhh good then, thanks ^^

closed 2020-03-26 by Dolu1990

aswaterman, 2020-03-26:
You're welcome!